lemmingDev/ESP32-BLE-Gamepad #84: MD5 do not match

Kean4711DUB, 2021-11-08:
Anytime I "include" the BLE gamepad library to the code the board won't flash. It only flashes without it and without it I can't use the library(Obviously) The error code is as follows, "A fatal error occurred: MD5 of file does not match data in flash!" Any suggestions would be greatly appreciated. Thanks in advance.

lemmingDev, 2021-11-08:
Weird

Is this with unmodified examples?

Are you using any of the pin numbers reserved for the internal flash?

Have you tried a different USB cable?

Kean4711DUB, 2021-11-08:
Thank you for the speedy response. Yes even with unmodified examples this happens and I tried to take the board off of any connections once with the same concern. I would add that I am using an ESP 32 wrover Dev Module that has a camera and wifi and Bluetooth built in that do pre-occupy some pins, but from my understanding they only use those pins if instructed to. I did try 2 different USB cables. I'm currently waiting on a different board to see if it has the same issue. Let me know what you think. If you want me to reverify anything or try something else.

Kean4711DUB, 2021-11-10:
The new board did not run into that issue.

lemmingDev, 2021-11-10:
Thanks for the heads up.

All ESP32 have Wi-Fi and Bluetooth built in and don't occupy pins, but it's weird that having a camera attached would prevent it from working.

All ESP32-S2 are Wi-Fi only, and don't have Bluetooth

If it was an ESP32-S2-WROVER --> they don't have Bluetooth, however the ESP32-WROVER do have it. I wonder if that was the issue?

In any case, I'm glad the new board has sorted it out.

Kean4711DUB, 2021-11-10:
Thank you! I don't believe it was an s2 but I am getting used to this still. and I misunderstood at first. the camera can preoccupy not the wifi and bluetooth as you mentioned. Also weird quirk. Pin 35 is either flickering or always on but other pins work fine for the gamepad buttons. Any ideas on why? is there certain pins that can't use buttons or pull-up resistors in the esp32?

lemmingDev, 2021-11-10:
Yeah - pins 35~39 are input only and are not connected to input pullup resistors

lemmingDev, 2021-11-10:
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

closed 2021-11-11 by lemmingDev