lemmingDev/ESP32-BLE-Gamepad #85: ESP32-BLE-Gamepad sending/update battery level

LuXeLLo, 2021-11-24:
Good morning

At the outset, I would like to point out that I am a beginner and I do not know programming very well, I am learning the basics all the time

I use the ESP32-BLE-Gamepad library and the https://github.com/pangodream/18650CL library to calculate the battery level,

I added bleGamepad.setBatteryLevel to the code.

The problem is that in windows 10 the battery level is only read when adding the device. To update the battery level I need to remove and add the device, I want to get battery update every 2 minutes

here is the code:

#define BOUNCE_WITH_PROMPT_DETECTION    // Make button state changes available immediately

#include <Bounce2.h>      // https://github.com/thomasfredericks/Bounce2
#include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad
#include <Pangodream_18650_CL.h>

#define numOfButtons 14

Bounce debouncers[numOfButtons];
Pangodream_18650_CL BL;

byte buttonPins[numOfButtons] = { 4, 5, 13, 14, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33 };
byte physicalButtons[numOfButtons] = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14 };

const unsigned long eventInterval = 120000;
unsigned long previousTime = 0;

void setup()
{
  for (byte currentPinIndex = 0 ; currentPinIndex < numOfButtons ; currentPinIndex++)
  {
    pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
    
    debouncers[currentPinIndex] = Bounce();
    debouncers[currentPinIndex].attach(buttonPins[currentPinIndex]);      // After setting up the button, setup the Bounce instance :
    debouncers[currentPinIndex].interval(5);        
  }

  bleGamepad.begin(numOfButtons);
  bleGamepad.setAutoReport(false);
  Serial.begin(115200);
}

void loop()
{
  if(bleGamepad.isConnected())
  {
    bool sendReport = false;

    unsigned long currentTime = millis();
    
    if (currentTime - previousTime >= eventInterval) 
    {
      bleGamepad.setBatteryLevel(BL.getBatteryChargeLevel());
      sendReport = true;
      Serial.print("Charge level: ");
      Serial.println(BL.getBatteryChargeLevel());
    
      /* Update the timing for the next time around */
      previousTime = currentTime;
    }    
    
    for (byte currentIndex = 0 ; currentIndex < numOfButtons ; currentIndex++)
    {
      debouncers[currentIndex].update();

      if (debouncers[currentIndex].fell())
      {
        bleGamepad.press(physicalButtons[currentIndex]);
        sendReport = true;
        Serial.print("Button ");
        Serial.print(physicalButtons[currentIndex]);
        Serial.println(" pushed.");
      }
      else if (debouncers[currentIndex].rose())
      {
        bleGamepad.release(physicalButtons[currentIndex]);
        sendReport = true;
        Serial.print("Button ");
        Serial.print(physicalButtons[currentIndex]);
        Serial.println(" released.");
      }
    } 
    
    if(sendReport)
    {
      bleGamepad.sendReport();
    }

    //delay(20);	// (Un)comment to remove/add delay between loops 
  }
}

Thank You for help and sorry for the mistakes, I used a translator

lemmingDev, 2021-11-25:
Hi

I'd never actually tried that functionality.

This library is a modification of a mouse library, and I've just tested that library and this library and neither of them seem have the setBatteryLevel function working

I've just posted a question on that Github page

I'll update this post when I have more info

lemmingDev, 2021-11-27:
Hi

I heard back from the creator of the BLE mouse library.
He confirms that the setBatteryLevel function doesn't work after the device connects.

You can only set it when it connects

I'll update the readme in the next release

closed 2021-11-27 by lemmingDev

LuXeLLo, 2021-11-27:
Hi,
Ok, Thank you for answer.