lemmingDev/ESP32-BLE-Gamepad #86: > i have 4.0, 4.2 and 5.0.

pipech90, 2021-12-13:
> Hi!
> 
> I have a problem. I am trying to make a gamepad for pc. But the PC does not see it in a bluetooth environment. My phone sees and connects. I tried to find it using Windows 8.1 and 10. What could be the problem?

You need at least Bluetooth 4.0...

_Originally posted by @lemmingDev in https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/58#issuecomment-992481623_

lemmingDev, 2021-12-13:
Hi

Not sure why it wouldn't work. Have had it working on dozens of Windows PCs without issue
Have you got any other Windows PCs to try it with.
Could be a software / driver issue

pipech90, 2021-12-14:
lemmingDev, thanks, found another computer.  He sees the device.  But a second after connecting, it disconnects.

closed 2021-12-14 by pipech90