lemmingDev/ESP32-BLE-Gamepad #87: Custom PID - suggestion

TheRussian007, 2022-01-01:
Hi,

I am using 3 identical ESP32 modules as gamepads. All are reporting VID_E502&PID_BBAB and will have the same Game Controller name because of that. I've tried changing the name (posted here a few months back), but only can do so in RegEdit. However as it is only one entry for the same VID and PID, it will change the name for all 3 modules to always the same.
This causes issues in games as they sometimes mix up the modules and I have to reassign all buttons again.

I have fond a solution by adding a pid option to your BleGamepad object in my project, like this:
`BleGamepad bleGamepad("Hydro Sequential", "turbo-labs", 100, 0xacbb);` <-- I can define the PID here now

I have added an additional attribute into BleGamepad.cpp and BleGamepad.h to make it work.
Ultimately it is this line in BleGamepad.cpp that makes it work for me:
`BleGamepadInstance->hid->pnp(0x01,0x02e5,BleGamepadInstance->pid,0x0110);`

Described in NimBLE:
[image]

The hex value [0xabbb] in BleGamepad.h is later shifted in by one byte in pnp of the NimBLE library, resulting in PID_BBAB.

As end result I have 3 ESP32 Gamepads with unique naming (still configured in regedit), by having unique PIDs.

[image]

Would be really handy if you could add this as an option into your library. Makes a big difference for someone who is using multiple similar devices.
Thank you. Great library btw.

holography-be, 2022-01-27:
I agree, it should be a nice improvement.

lemmingDev, 2022-02-02:
Thanks

Will do...

holography-be, 2022-05-05:
Thanks a lot !

closed 2022-05-26 by lemmingDev