lemmingDev/ESP32-BLE-Gamepad #88: Can only see/connect to bluetooth device within a second of boot

redsil, 2022-01-31:
I am not sure what could be causing this.  The gamepad library, the nimble library, something else with my esp32 devkit?
Windows seems to be slow at scanning so it won't even see the device most of the time.  My android phone will see it immediately after resetting the esp32, but if I try scanning for devices much after a few seconds after reset, it won't appear.  Also, I can't pair with the esp32 unless it does it within that very small window after reset.  After it is paired, it remains that way and seems to function correctly as a gamepad. 

Using:
ESP32_BLE_Gamepad-0.3.4
NimBLE_Arduino-1.3.6

O/S is Linux, using arduino-cli to compile and upload

Here is a small program that exhibits the behavior for me:

#include <BleGamepad.h> 

O/S is Linux, using arduino-cli to compile and upload

Here is a small program that exhibits the behavior for me:

#include <BleGamepad.h> 

BleGamepad bleGamepad("GPTester","Nobody",90);

void setup(){
  Serial.begin(115200);  
  Serial.println("Starting BLE work!");

  bleGamepad.setAutoReport(false);
  bleGamepad.begin(32, 0,false,
    false,
    false,
    false,
    false,
    false,
    false,
    false);  
 
}

void loop(){
  delay(50);
}

redsil, 2022-01-31:
Ok, I was able to resolve this.  Hopefully if someone else encounters this, they will stumble up on this and find it helpful.
I only needed to erase the flash memory.  Once I did that and uploaded the sketch again, it was back to working like I expected.
esptool.py --port <port> --chip auto erase_flash

Also note that my linux dist had esptool v 2.8 installed, but I needed a newer one (3.2), otherwise it would say erase_flash was not supported.

I would love to know what in flash memory was affecting bluetooth.

lemmingDev, 2022-02-02:
Glad you got it sorted...

closed 2022-05-26 by lemmingDev