lemmingDev/ESP32-BLE-Gamepad #89: feature request: external led indicator status

alexdelli, 2022-02-01:
hi, is it possibible to connect a simple led to check the status of ESP32? (on/off simply)

lemmingDev, 2022-02-02:
Wouldn't you just set a pin connected to an LED high?

It would light when there is power, and turn off when there is none...

holography-be, 2022-02-02:
a way to do this

#include <Arduino.h>
#include <bleGamepad.h> 

#define LedConnect 13

ulong elapsed = 0;
bool bleWaiting = false;

BleGamepad ble();

void waitConnection()
{
  if(elapsed < millis())
  {
    elapsed = millis() + 500; // blink 0.5 second
    digitalWrite(LedConnect,!digitalRead(LedConnect));
  }  
  bleWaiting = true;    
}

void setup()
{
  Serial.begin(115200);
  pinMode(LedConnect, OUTPUT);
  digitalWrite(LedConnect, LOW); // Led OFF 
  ble.setAutoReport(false);
  ble.setControllerType(CONTROLLER_TYPE_GAMEPAD);
  ble.begin(32,1);

  Serial.println("Waiting connection  ");   
  elapsed = 0;
  while(!ble.isConnected())
  {
    waitConnection();    // 
  }
  bleWaiting = false;
  digitalWrite(LedConnect, HIGH);
  Serial.println("Connected");
}

void loop()
{
    if(ble.isconnect())
    {
        if(bleWaiting)  // lost and recovered connection
        {
            digitalWrite(LedConnect, HIGH);   
            bleWaiting = false;         
        }
        //
        // stuff to do ...
        //
        //
    }
    else
    {
        waitConnection();
    }
}

holography-be, 2022-02-02:
void setup()
{
   pinMode(13, OUTPUT);  // led connected on GPIO13
   digitalWrite(13, HIGH);
}

holography-be, 2022-02-02:
and if you wish to know the status of BleGamepad connection.
Led is ON when connected, blink if not.

#include <Arduino.h>
#include <BleGamepad.h> 

#define LedConnect 13

ulong elapsed = 0;
bool bleWaiting = false;

BleGamepad ble();

void waitConnection()
{
  if(elapsed < millis())
  {
    elapsed = millis() + 500; // blink 0.5 second
    digitalWrite(LedConnect,!digitalRead(LedConnect));
  }  
  bleWaiting = true;    
}

void setup()
{
  Serial.begin(115200);
  pinMode(LedConnect, OUTPUT);
  digitalWrite(LedConnect, LOW); // Led OFF 
  ble.setAutoReport(false);
  ble.setControllerType(CONTROLLER_TYPE_GAMEPAD);
  ble.begin(32,1);

  digitalWrite(LedConnect,HIGH);
  Serial.println("Waiting connection  ");   
  elapsed = 0;
  while(!ble.isConnected())
  {
    waitConnection();    // 
  }
  bleWaiting = false;
  digitalWrite(LedConnect, HIGH);
  Serial.println("Connected");
}

void loop()
{
    if(ble.isconnect())
    {
        if(bleWaiting)  // lost and recovered connection
        {
            digitalWrite(LedConnect, HIGH);   
            bleWaiting = false;         
        }
        //
        // stuff to do ...
        //
        //
    }
    else
    {
        waitConnection();
    }
}

alexdelli, 2022-02-02:
yeeeeeeeessssssss!!!!
Perfect!
Thank you very much

About battery status? Does exist a sketch?

lemmingDev, 2022-02-22:
The battery status can only be set each time it connects, and not throughout the course of the sketch. (Bug?)

So, you could check your battery status when the device is powered on, and set it correctly when you create the gamepad object, but then as the battery depletes while you're using the device, you're unable to update the battery status percentage to the host...

You could however add a battery warning LED and set it to turn on when it gets to about 15 percent, and then have it start blinking at 10% and get faster as it gets closer to running out etc..

Google turns up a few results for working out the battery percentage on ESP32s

alexdelli, 2022-03-17:
i post some corrections:

`
#include <BleGamepad.h>       // https://github.com/MagnusThome/ESP32-BLE-Gamepad
#include <Arduino.h>

BleGamepad bleGamepad("ESP32", "Espressif", 100);

#define LedConnect 13

ulong elapsed = 0;
bool bleWaiting = false;

void waitConnection()
{
if(elapsed < millis())
{
elapsed = millis() + 500; // blink 0.5 second
digitalWrite(LedConnect,!digitalRead(LedConnect));
}
bleWaiting = true;
}

void setup() 
{
pinMode(13, OUTPUT); // led connected on GPIO13
digitalWrite(13, HIGH);

  Serial.begin(115200);

pinMode(LedConnect, OUTPUT);
digitalWrite(LedConnect, LOW); // Led OFF
bleGamepad.setAutoReport(false);
bleGamepad.setControllerType(CONTROLLER_TYPE_GAMEPAD);
bleGamepad.begin(32,1);

digitalWrite(LedConnect,HIGH);
Serial.println("Waiting connection ");
elapsed = 0;
while(!bleGamepad.isConnected())
{
waitConnection(); //
}
bleWaiting = false;
digitalWrite(LedConnect, HIGH);
Serial.println("Connected");
}

void loop() {

if(bleGamepad.isConnected())
{
if(bleWaiting) // lost and recovered connection
{
digitalWrite(LedConnect, HIGH);
bleWaiting = false;
}
//
// stuff to do ...
//
//
}
else
{
waitConnection();
}
}

`

closed 2022-03-17 by alexdelli