lemmingDev/ESP32-BLE-Gamepad #90: Gamepad issues on ESP32-S3-Devkit-C

digitalLumberjack, 2022-02-20:
Hello o/

I'm a newcomer in the ESP world. So thank you very much for making this easier with great libs like this one!

I have an ESP32-S3-Devkit-C and i'm having the ESP32-BLE-Gamepad running on it. It works on ubuntu, but I have some questions.

**Issue 1**

On Ubuntu 21.10, with a kernel 5.13, **the gamepad is working well** ! 
But when i connect the gamepad, i have this log in dmesg :
```
[  421.572410] Bluetooth: hci0: unexpected SMP command 0x07 from 7c:df:a1:e2:94:a5
[  422.158127] input: Controller as /devices/virtual/misc/uhid/0005:E502:BBAB.000C/input/input44
[  422.158438] hid-generic 0005:E502:BBAB.000C: input,hidraw6: BLUETOOTH HID v10.01 Gamepad [Controller] on 9c:b6:d0:91:36:b2
```

------

**Issue 2**

On recalbox with a 5.10 kernel, the controller is **paired successfully, and I can connect it**.

But **I have no joystick detected in /dev/input/jsX**, and the only logs in dmesg are :
```
[  315.915746] Bluetooth: hci0: security requested but not available
[  321.179711] Bluetooth: hci0: Malicious advertising data. Stopping processing
[  321.180364] Bluetooth: hci0: Malicious advertising data. Stopping processing
```

In bluetoothctl
```
[bluetooth]# info 7C:DF:A1:E2:94:A5
Device 7C:DF:A1:E2:94:A5 (public)
	Name: Controller 
	Alias: Controller 
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:vE502pBBABd1001
	Battery Percentage: 0x64 (100)
```

Do you have any idea that could help me ?

---------
For the context I build my firmware with this platform.io configuration:

```
[env:esp32s3]
board = esp32-s3-devkitc-1
platform = https://github.com/digitallumberjack/platform-espressif32.git#f3f0eeb273760b80740949e840dc8bf90b5c2255
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/674/framework-arduinoespressif32-v4.4_dev-e8af343d93.tar.gz
framework = arduino
lib_deps =
    ESP32-BLE-Gamepad@0.3.4
    NimBLE-Arduino=https://github.com/h2zero/NimBLE-Arduino.git#658339b8fdee81c644608ea898a7294c2275574b
monitor_speed = 115200
board_build.partitions = noota_ffat.csv
board_build.filesystem = ffat
extra_scripts = replace_fat.py
```

lemmingDev, 2022-02-22:
Hi

No idea I'm afraid

Have only really tested in on Windows personally. Other users reported the compatibility for the other OSs listed.

If you nut it out, let me know...

davidgrohl, 2022-03-21:
@digitalLumberjack, did you try controller with retropie?

digitalLumberjack, 2022-03-21:
Hi, not yet, but we bumped the kernel at recalbox for the next version, i'll see if it fixes anything.

lemmingDev, 2022-05-26:
Will close this as its working well with Ubuntu 21.10, with a kernel 5.13

closed 2022-05-26 by lemmingDev