lemmingDev/ESP32-BLE-Gamepad #91: Button mapping issues

DoomyDoomer, 2022-02-21:
This is an outstanding project and I've picked up the basics of how it works but I'm having some issues with button mapping. I'm trying to take an old controller and map each button to match what windows or other bluetooth adapters would recognize an Xbox One / Series S/X controller without having to remap with software. I'm going to use Neo Geo as my mapping example below.

**My button mapping:** 
XBOX -----> Neo Geo("physicalButtons")
A  ----------> A (1)
B  ----------> B (2)
Y  ----------> D (3)
X  ----------> C (4)
View -------> Select (5)
Menu ------> Start (6)
Up ---------> Up (Assigned to Hat 1)
Left --------> Left (Assigned to Hat 1)
Down ------> Down (Assigned to Hat 1)
Right -------> Right (Assigned to Hat 1)

_Share Btn, R1, R3, L1, L3, L and R analog sticks and triggers are all unmapped_

**My mapping issues:**
A & B work as they should. 
D (3) is mapped to Xbox X and C is mapped to the Xbox Share Btn. 
Select is mapped to Xbox Y and Start seems to be mapped to a button out of range. 
Lastly, the DPAD is interesting because in some testing apps the work but the Left and Right are reversed. In others, the mapped dpad hats aren't recognized as the xbox dpad. Hopefully that made sense.

**A snipped of my sketch:**
```
#define numOfButtons 6
#define numOfHats 1  // Maximum 4 hat switches supported

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = { 4, 0, 18, 1, 10, 19 }; 
byte physicalButtons[numOfButtons] = { 1, 2, 3, 4, 5, 6 }; 

byte previousHatStates[numOfHats*4];
byte currentHatStates[numOfHats*4];
byte hatPins[numOfHats*4] = { 6, 2, 3, 5 };
```

I've made sure to check the obvious like mismatching GPIO pin connections but those are all correct. I'm using the suggested controller tester program as well as various other testers online. I also have a ESP32-WROOM running a custom controller tester firmware that is probably the most accurate since it reads out all the xbox buttons and lights them green when any given button is pressed.

lemmingDev, 2022-02-22:
Hi

My advice would be to remove the old controller from the equation to start, and wire up standard buttons (or even some wires to manually touch) directly for testing. 

Once you've got that working, then reintroduce the old controller. 

My guess is that as the old controller you're trying to use still has the buttons wire up to circuitry that is messing stuff up.

FYI - On Windows, I've tried to map buttons to what an Xbox 360 controller would be

Let me know how you go

lemmingDev, 2022-02-22:
Oh - and I think Pin 1 is the TX pin, so use another one

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

lemmingDev, 2022-02-22:
And pin 6 is used for the SPI flash, so don't use that one either

DoomyDoomer, 2022-02-22:
Thanks for your quick reply. I forgot to mention that I'm using an ESP32-C3-Mini, so it's a different pinout to the ESP32-WROOM and I don't believe I'm using any pins that would cause any issues. The Neo Geo controller is basically a simple button press with no line buffers or microcontrollers under the hood, so it should be a perfect controller to use as a first step. 

Maybe due to the fact that you've set this up as a 360 controller is the issue I'm having. If I were to align the code to an Xbox Series X controller, where would I find that info within the src?

lemmingDev, 2022-02-22:
Hi

https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/BleGamepad.cpp

The HID Report Descriptor is configurable, so the user can specify what elements of the controller are enabled when they create the gamepad object. By default, it sets up a gamepad as close to xbox360 as possible, though it's not detected as one as the Vendor and Product IDs are different, plus it doesn't have the proprietary security chip. The user can even set it up as a racing simulator wheel or flight stick...

BTW - the HID my library produces doesn't explicitly set a select and start button, for example, but relies on the host operating system to determine that certain buttons (7 and 8 from memory, but could be some other values) are the select and start. From what I gathered from HID report descriptors floating around on the net, this is how most are configured.

I would still remove the controller from the equation temporarily to confirm it isn't interfering

lemmingDev, 2022-05-05:
The new version 4.0 of the library can now explicitly create HID start and select buttons (enabled by default)

Will add two additional buttons

Second-last button is select, last button is start

closed 2022-05-26 by lemmingDev