lemmingDev/ESP32-BLE-Gamepad #92: Problem compiling in Arduino Ide

davidgrohl, 2022-03-30:
Hi,
I'm using an ESP32-S2 devkit and I'm working on Arduino IDE. I installed NimBLE-Arduino library and ESP32-BLE-Gamepad-master library. I added ESP32 support and selected ESP32-S2 devkit board. The problem is that I cannot compile, IDE give me error "fatal error: esp_bt.h: No such file or directory". Any suggestions?

lemmingDev, 2022-03-30:
Hi

As far as I'm aware, the  ESP32-S2 does not have Bluetooth...

It does have USB HID support though, so theoretically, someone could make a USB gamepad library for it

lemmingDev, 2022-03-31:
Seems there is already a great library for the ESP32-S2 built in

[image]

davidgrohl, 2022-03-31:
Thank you very much mate. I bought an ESP32 and I'll give it a try. I'll come back to you to get help :)

davidgrohl, 2022-03-31:
I installed ESP32 support, NimBLE-Arduino and ESP32-BLE-Gamepad libraries but when I continue to get error when I try to compile an example.

[image]

[image]

And this is the log (I enabled verbose logging).
`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\R&D\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\R&D\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\R&D\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path C:\Users\R&D\AppData\Local\Temp\arduino_build_858171 -warnings=all -build-cache C:\Users\R&D\AppData\Local\Temp\arduino_cache_13479 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\R&D\Documents\Arduino\libraries\ESP32-BLE-Gamepad\examples\Gamepad\Gamepad.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\R&D\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\R&D\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\R&D\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path C:\Users\R&D\AppData\Local\Temp\arduino_build_858171 -warnings=all -build-cache C:\Users\R&D\AppData\Local\Temp\arduino_cache_13479 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\R&D\Documents\Arduino\libraries\ESP32-BLE-Gamepad\examples\Gamepad\Gamepad.ino
Using board 'esp32' from platform in folder: C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "C:\\Users\\R&D\\Documents\\Arduino\\libraries\\ESP32-BLE-Gamepad\\examples\\Gamepad\\partitions.csv" copy /y "C:\\Users\\R&D\\Documents\\Arduino\\libraries\\ESP32-BLE-Gamepad\\examples\\Gamepad\\partitions.csv" "C:\\Users\\R&D\\AppData\\Local\\Temp\\arduino_build_858171\\partitions.csv"
& non atteso.
exit status 1
Error compiling for board ESP32 Dev Module.
`

Any suggestion?

closed 2022-03-31 by davidgrohl