lemmingDev/ESP32-BLE-Gamepad #93: Feature Request: Add a way of disconnecting and advertising the bluetooth device without a power cycle

CodingMommy, 2022-03-31:
Would it be possible to add a function that when called disconnects from the current device and advertises the bluetooth device again so it can be connected to? I'm currently trying to get my head around the nimble library to see if I could offer a way of implementing it but I'm not really sure.

Thank you

LeeNX, 2024-11-23:
I am sure this would be a supported function of the underlying BlueTooth stack. Should look at their docs, but a working example might be useful.  
  
Think that possible a way to ban or accept but repair to another device would be nice feature. Not sure how often people re-pair their gamepads, but something I would use.

lemmingDev, 2025-01-29:
Long standing issue now addressed from v0.6.9 and above

closed 2025-01-29 by lemmingDev