lemmingDev/ESP32-BLE-Gamepad #94: Windows 11 not recognizing as gamepad

cornem, 2022-04-11:
A couple of weeks ago I started experimenting with an ESP32 board and this library. The first time I tried it out, it worked great. Windows 11 paired with the device and it showed up as a gamepad.

Since then I actually finished the other parts of the project and put it all together. I had to change some of the code and uploaded it to the device.

However, Windows no longer recognizes it as a gamepad. I've 'removed' the device while connected and tried to reinstall it. Now, when the `Add device` popup scans for Bluetooth devices, the gamepad shows up and also has a gamepad icon. However, if I then select it, it installs it but it no longer has a gamepad icon. If I go to the 'old' control panel in Windows, it actually lists it as an unknown device. While in this state, the `gamepad.isConnected()` returns `true`, and it's sending reports.

Interestingly, even though I've uninstalled the device from Device Manager, when I reconnect it and look at the device history, it goes back to the very first time I connected to it (when it worked).

Any ideas? Any ideas on how to fully remove a device from Windows so it completely forgets about it?

lemmingDev, 2022-05-05:
Weird. Haven't seen that before

New version 4.0 added. Give it a crack

A new feature of the library is to configure a custom VID and PID, so that may help to have Windows think it is a new device.

Look at TestAll.ino for an example

lemmingDev, 2022-05-05:
Also, I wonder if erasing the flash memory may help

https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/88

cornem, 2022-05-06:
Erasing the flash memory made no difference. What eventually fixed it was to change the controller type - I changed it to a Joystick, and then it worked again. Closing for now - I suspect it's something with Windows, not so much this library. However, since I only had 1 Windows PC, I couldn't test it against other devices when it didn't work.

closed 2022-05-06 by cornem

lemmingDev, 2022-05-07:
Thanks for the update. Glad you got it working in the end