lemmingDev/ESP32-BLE-Gamepad #99: Is it possible to implement force feedback?

David1r20, 2022-05-03:
Is there a possibility to get ffb?

lemmingDev, 2022-05-05:
Hi

Was asked a couple of years ago

https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/12

Out of my area of expertise

If someone wants to take a crack at it and create a pull request, it would be greatly appreciated.

Narsku, 2022-05-22:
Sorry for my poor english. with ffb do you mean a simple vibrator motor?, i found an app for android that test the gamepad and the vibration, can you catch this command and implement it to the library? or this is the complicated part?

lemmingDev, 2022-05-23:
Sorry.
I have no idea how to get it working...

LeeNX, 2022-05-23:
Force feedback or haptic is possible, but not yet quite standard, at least not in my very basic research.

There are a few options:
 1. make your own interface for your project ie: BLE serial - https://www.hackster.io/chuartdo/ble-haptic-dual-joystick-controller-31c2c2 
 2. emulate one of the other vendors haptic system XBox/PSX controller (not sure the legal status) - 
 3. setup the correct HID and output fields with callbacks, like CapsLock LED - https://github.com/tloimu/adapt-ffb-joy/tree/wiki
 4. Possible part of 3 - Extend support for FFB in SDL2 - https://github.com/libsdl-org/SDL/tree/main/src/joystick/hidapi

3 and 4 are something in my spare time I am trying to understand and added, but it's slow going and just a little over my current skill level.

Hope this might help somebody else just start their progress.

Narsku, 2022-05-23:
Ok!, I'm gonna try to work on it

lemmingDev, 2022-05-27:
That would be great.
Let me know if you make any progress

kotobuki, 2022-06-25:
Let me jump into this issue because I'm also interested in it.

@Narsku 
https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/99#issuecomment-1134022276
> i found an app for android that test the gamepad and the vibration, can you catch this command and implement it to the library? or this is the complicated part?

Which app did you try? I also want to try the app.

lemmingDev, 2022-09-20:
Seems some of the info needed is here
https://www.usb.org/sites/default/files/pid1_01_0.pdf

lemmingDev, 2022-09-20:
And here's an Arduino Leonado joystick library with force feedback
https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

LeeNX, 2022-11-19:
I have been reading through the [ArduinoJoystickWithFFBLibrary](https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary) repo and came across https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary/issues/30#issuecomment-870345902 where extending the HID PID seems to get FFB working in Linux, so I was thinking of extending ESP32-BLE-Gamepad PID to include the changes for test - https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary/blob/master/src/FFBDescriptor.h 
  
but after about a 1/3 of the way through, worried about any typos or other problems, hoping there might be a quicker or even better way to do this, for just a test? @lemmingDev you have an idea? My fork and branch in it's current state - https://github.com/LeeNX/ESP32-BLE-Gamepad/commits/leet-ffb-pid 
  
 Also thinking, if we going to use this code, we should possible ask or inform https://github.com/YukMingLaw that we doing this? Just good citizenship ... I will reach out, if nobody else beats me too it.

MatthewReed303, 2022-12-01:
Hi @LeeNX 

Have you had any luck getting the FFB to work? I was looking at using ESP32-S2 and convert the [ArduinoJoystickWithFFBLibrary](https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary) to work with the esp32, but if this works that's even better.

Thanks

LeeNX, 2022-12-02:
@MatthewReed303 still in progress, have not made much headway since my last post. Not much feedback either, so I am not sure if anybody else would find this useful. I am still keen and interested and will come back to this. Just need to find the time and motivation. Have some free time coming up in two weeks, so I think I will be able to get to at least testing PID with OSX and Linux (Raspberry Pi) with SDL 2/3. 
  
Been hacking at a simple console testing program, to test some of my ideas. Was very interested in seeing very much different feature reporting back on Linux vs OSX. On Linux, if I set SDL hint that my BLE devices is PS4 controller, not much changes, but on OSX, when hinted as PS4 controller, the serial number from BLE is reported back differently. SDL feature (bug) that I was not expecting. I don't even remember why I did think of even trying this.  
  
I hope there is more interest, which would be motivating for me.

tapir, 2022-12-24:
Does this library have a callback for messages received from the PC? I'm trying to do some latency measurements and my idea was to send FFB messages to the controller and respond back with a button press thus having the full round-trip latency.

gagagu, 2023-07-11:
I'm using the [Fino](https://github.com/jmriego/Fino] library which based on the [YukMingLaw](https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary) to create an [Arduino FFB Yoke](https://github.com/gagagu/Arduino_FFB_Yoke) for FlightSim. 
I'm thinking about to make a new design of the yoke to make it smaller and may be use an ESP32 processor to bring my project to a next level (Bluetooth, faster, wifi). 

Is there any progress of implementation of FFB? 
You would make may people very happy for implement, especially the Flightsim and RacinSim Guys!

ATUK-OoD, 2025-05-29:
Hello, just wanted to ask, is there any progress on this?

Narsku, 2025-05-30:
> Hello, just wanted to ask, is there any progress on this?

Idk but some time ago i found a fork of this repository

https://github.com/Mystfit/ESP32-BLE-CompositeHID

In my test the rumble only works with steam input

ATUK-OoD, 2025-05-30:
well, that's a shame there is only rumble. I need normal force feedback.

fazerlab, 2025-08-01:
First, thank you for this cool project.
I find this project that emulate G27 Wheel and add a descriptor for FFB ( https://github.com/michal2229/g27_emu) .
I craft a code and could emulate controls and receive FFB messages. The code and configurations are here: https://gist.github.com/fazerlab/23d758c7a9821916d99a70b106793914

The example  https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/master/examples/TestReceivingOutputReport
 could be a path to to add FFB?
The example TestReceivingOutputReport could be extended to receive messages FFB from PC?

My understand is that need a descriptor and functions to read reports. I don´t have knowledge to do this but could help with tests in esp32-S3 and references.

fazerlab, 2025-08-02:
Hi @LeeNX , you are right there is no documentation, but with some investigation i think that is a match of VID/PID ,  HID descriptor setup to allow receive the messages and app (game) configuration. For example if game detect a Logitech wheel it will send correct messages to device. I will do more investigation and return , thank you.

TTXL3099, 2026-01-02:
Is there any progress?I look forward to this matter being realized.

LeeNX, 2026-08-27:
There are a number of ways this could be possible, but it is totally up to the host OS to implement or a customer driver tool on the host OS for host level support.

There too is custom application support, something similar to SInput support in SDL3, take a look at #288 and PR #337 , which has PlayerLED working, but this only works with SDL3 via HIDApi, which I have only tested on Linux. I will work further on PR #337 and try get the SInput rumble working, not sure about the haptic feedback, but there is progress.

If anybody would like to jump in and take a look at PR #337 , test and give feedback would be helpful.

Things that might be worth putting more info into, is possible documenting Host OS and their ForceFeedback/rumble/haptic/etc services and protocols used and supported. Maybe info on alternatives, like [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID) which is basically an XBox with rumble support, this is already in the repo ReadMe file.

I think this issue is close to be covered as best as possible at the moment and could possible be closed.

LeeNX, 2026-09-18:
As of version v0.8.0, I believe this is covered, in at least two different ways, if not four, but I think the second two are a little less standard.

I think we can close this issue?

lemmingDev, 2026-09-18:
Another one to (finally) tick off the list
Thanks LeeNX

Closed!

closed 2026-09-18 by lemmingDev