lemmingDev/ESP32-BLE-Gamepad #102: Restart for version 4.0

jisle15064468204, 2022-05-22:
Download the gamepad instance to the development board and connect the Bluetooth development board to restart!

lemmingDev, 2022-05-23:
Hi
Please give me the exact code you are trying to run so I can test it
Thanks

jisle15064468204, 2022-05-26:
I use platformio download have restart occure,howevet used Arduino IDE is ok.

lemmingDev, 2022-05-26:
I do not use PlatformIO so I am unable to provide support on that

The user @dexterdy that submitted the pull request for most of the 4.0 changes used PlatformIO. Perhaps he can chime in

jisle15064468204, 2022-05-27:
Ok Thanks very much!

lemmingDev, 2022-05-27:
So, I got VS Code/PlatformIO installed and Arduino up and running. I compiled a basic gamepad sketch and was able to see it go into a reboot loop a couple of seconds after each time it makes a BT connection

Hopefully dexterdy can take a look

lemmingDev, 2022-07-29:
I had another crack at using VS Code/Platform IO and I was able to get it working perfectly this time

closed 2022-07-29 by lemmingDev