lemmingDev/ESP32-BLE-Gamepad #104: Clarification / Feature Request

GamingNJncos, 2022-06-03:
Requesting some input on where to isolate and change he MAX/MIN Descriptor values. Throughout the library -32767 is configured as a min value however it's unclear where that is specifically defined. 

I believe they originate from somewhere inside this cpp file but may be pulled from the int type directly. 
https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/d24c872f0e2f5327be95fb260745ce8bcb76e0d8/BleGamepad.cpp

Potential Feature request: 
Ability to manually define max/min per type (axes, trigger, etc).  

The request originates from situations where a connected device may not correctly understand negative values as a range (ie. signed int cast to unsigned value processing HID/HCI info) in these use cases working in purely "positive" values will permit the library to be compatible.  

A secondary benefit would arise in the form an input device that only offers fixed values/steps (Device returns only 0-15 for example). While possible to work-around leveraging things such as MAP being able to maintain and define 1:1 with a device would be desirable.

Looking at a number off off-shelf devices such as the Xbox One X/S and PS4 negative values are never used.

PS5: 
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)

Xbox: 
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)

lemmingDev, 2022-06-03:
Hey

[image]

0x7FFF for positive

and 

0x0801 for negative

Let me know if you need a hand

lemmingDev, 2022-08-20:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.5.0

v5.0 includes the ability to set min and max values for axes

closed 2022-08-20 by lemmingDev