lemmingDev/ESP32-BLE-Gamepad #106: How to increase the button?

mbohben, 2022-06-10:
sorry im new to micro controller programing, ok i'll be direct.

how to remove axis? and just set the number of button matrix, tried edit the main.cpp but seems nothing work on me.

tried to edit the BLEgamepad.h and blegamepadconfig too, and still didnt get it to work.

````
#include` <Arduino.h>
#include <keypad.h>
#include <BleGamepad.h>

BleGamepad bleGamepad("ESP32wheel", "lemmingDev", 100); // Shows how you can customise the device name, manufacturer name and initial battery level

#define ROWS 5
#define COLS 4
uint8_t rowPins[ROWS] = {13, 12, 14, 27, 26}; // ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {15, 2, 4, 16}; // ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
uint8_t keymap[ROWS][COLS] =
    {
        {1, 2, 3, 4},    // Buttons  1,  2,  3,  4      --> Used for calulating the bitmask for sending to the library
        {5, 6, 7, 8},    // Buttons  5,  6,  7,  8      --> Adjust to suit which buttons you want the library to send
        {9, 10, 11, 12}, // Buttons  9, 10, 11, 12      -->
        {13, 14, 15, 16}, // Buttons 13, 14, 15, 16      --> Eg. The value 12 in the array refers to button 12
        {17, 18, 19, 20}
};

Keypad customKeypad = Keypad(makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

///////////////////////////////////////////////////////////////////

void setup()
{
    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(false); // Disable auto reports --> You then need to force HID updates with bleGamepad.sendReport()
    bleGamepadConfig.setButtonCount(32);
    bleGamepad.begin();                    // Begin library with default buttons/hats/axes
}
````

DenisKozachok, 2022-06-11:
Hello)

You can add the next code.

`#define enableX false
#define enableY false
#define enableZ false
#define enableRX false
#define enableRY false
#define enableRZ false
#define enableSlider1 false
#define enableSlider2 false
#define enableRudder false
#define enableThrottle false
#define enableAccelerator false
#define enableBrake false
#define enableSteering false`

bleGamepadConfig.setWhichAxes(enableX, enableY, enableZ, enableRX, enableRY, enableRZ, enableSlider1, enableSlider2);      // Can also be done per-axis individually. All are true by default
  bleGamepadConfig.setWhichSimulationControls(enableRudder, enableThrottle, enableAccelerator, enableBrake, enableSteering); // Can also be done per-control individually. All are false by default
  bleGamepadConfig.setHatSwitchCount(numOfHatSwitches);                                                                      // 1 by default

DenisKozachok, 2022-06-11:
@mbohben How did you set up Visual Studio Code?
Could you please share the link to instruction)

Thanks)

mbohben, 2022-06-11:
> @mbohben How did you set up Visual Studio Code? Could you please share the link to instruction)
> 
> Thanks)

thanks for the tutorial above, will try it ASAP.
just install visual code right from the official web, then install platformio extension including c++ extension.

closed 2022-07-29 by lemmingDev