lemmingDev/ESP32-BLE-Gamepad #107: How to change device name?

DenisKozachok, 2022-06-11:
How to change the device name?

[image]

lemmingDev, 2022-06-14:
For now, has to be done in the registry I'm afraid

See here:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/77

spleen1981, 2022-07-26:
Can't fix this in the library I believe.
That name comes from a lookup list based on licensed VID&PID (e.g. [here](https://www.the-sz.com/products/usbid/index.php?v=0x045E&p=&n=)).
So there is no way to put your custom device name in that list unless you buy a license for your own VID/PID.

lemmingDev, 2022-07-26:
Gotcha - thanks for the explanation

At least there's a registry work around...

closed 2022-07-26 by lemmingDev