lemmingDev/ESP32-BLE-Gamepad #109: Turbo Functionality

DoomyDoomer, 2022-06-27:
I'm using ESP32-BLE-Gamepad to build a wireless PC Engine controller to connect to my PC Engine bluetooth adapter. I'm wondering if it's possible to implement turbo functionality and if so how I'd go about doing that?

lemmingDev, 2022-07-26:
You would just poll for the position of the switch at intervals and send the presses as needed

DoomyDoomer, 2022-07-27:
Thanks for getting back to me. Can you provide a code block example maybe with debounce included? Just to explain further the use case. I don't want the button to be in a turbo pressed state until the user presses the button. In other words, if the user has the switch for the "B" button set to turbo, the B button won't actually be firing rapidly until the user presses and holds the B button. This would be the same functionality as the original PC Engine controller.

lemmingDev, 2022-07-29:
Send though your current sketch

DoomyDoomer, 2022-07-29:
Below is my sketch. Right now my turbo functionality isn't in software except for the clock generation.  I'm using the same counter IC for turbo as the original PC Engine controller. The counter requires a clock value, basically a square wave input for timing. The original controller was fed 12hz from the console. I tried generating this via PWM but it's broken when using the Arduino IDE and the ESP32-C3-Mini. This is a shame because it should have been really simple to generate but I was able to confirm what others have posted using my oscilloscope, no PWM on the C3-mini. As a very ham fisted last ditch effort I have the clock running as a simple high/low loop with delay to generate the proper clock square wave signal. I have this running on the second core because it was obviously running too quickly and halting the other instructions on the first core. Even though this method works, I believe it could drain the battery faster, which isn't ideal.

All that being said, it would be great to be able to write the two steps of rapid fire ("slow" rapid fire, "fast" rapid fire) in software but I was unable to figure out how to get this working using the available examples. I tried using debounce but there wasn't a clear logical method to "sense" when the switch was active and rapid fire within a loop that's already firing off the press function and how the conditions are setup. Just something I couldn't wrap my head around, so any help is appreciated.

**Sketch (I use VLC):**

`#include <Arduino.h>
#include <BleGamepad.h>
#include <Bounce2.h>

void Task1code (void * pvParameters);
void Task2code (void * pvParameters);

TaskHandle_t Task1;
TaskHandle_t Task2;

// BLEGamePad Settings
BleGamepad bleGamepad("PCE2BT", "Humble Bazooka", 100);

#define LedConnect 0
#define numOfButtons 16
#define numOfHats 1
#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP 10        /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

ulong elapsed = 0;
ulong turboElapsed = 0;
unsigned long startMillis;  //some global variables available anywhere in the program
unsigned long currentMillis;
const unsigned long period = 1000;  //the value is a number of milliseconds
bool bleWaiting = false;

Bounce rapidFire = Bounce();

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = {2, 3, 8, 9}; // GPIOs
byte physicalButtons[numOfButtons] = {4, 1, 11, 12}; // II, I, SELECT, RUN

byte previousHatStates[numOfHats * 4];
byte currentHatStates[numOfHats * 4];
byte hatPins[numOfHats * 4] = {5, 7, 6, 4}; // in order UP, LEFT, DOWN, RIGHT. 4 pins per hat switch (Eg. List 12 pins if there are 3 hat switches)

void deepSleep()
{
  Serial.print("Going Into Sleep...");
  // to do
}

void waitConnection()
{
  if(elapsed < millis())
  {
    elapsed = millis() + 500; // blink 0.5 second
    digitalWrite(LedConnect,!digitalRead(LedConnect));
  }
  
  bleWaiting = true;
}

void setup()
{
  pinMode(18, OUTPUT);

  // Setup Buttons
  for (byte currentPinIndex = 0; currentPinIndex < numOfButtons; currentPinIndex++)
  {
    pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
[... 155 lines cut ...]

        delay(20);
    }
    else
    {
      waitConnection();
    }
  }
}

void Task2code( void * pvParameters )
{
  for(;;)
  {
    digitalWrite(18,HIGH);
    delay(12);
    digitalWrite(18,LOW);
    delay(12);
  }
}

void loop() {
 
}
`

closed 2024-02-24 by lemmingDev