lemmingDev/ESP32-BLE-Gamepad #110: Changing NimBLE con parameters after connection with BLE-Gamepad

GamingNJncos, 2022-06-28:
In an effort to potentially tune latency I have been looking at the Nimble library and found the updateConnParams option as discussed here:
https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_server.html#a9b08738766de90d7065263761314030a 

Also as shown in these Lines https://github.com/h2zero/esp-nimble-cpp/blob/master/src/NimBLEServer.cpp#L750-L762 

I believe this needs to be set via BleConnectionStatus.cpp for the BLE-Gamepad Library here: 

`void BleConnectionStatus::onConnect(NimBLEServer *pServer)
{
    this->connected = true;
     updateConnParams(desc->conn_handle, 0x10, 0x20, 0, 600); //example location
}`

Any idea if this is correct or should this be called from somewhere else? 
Documentation in Nimble notes "Can only be used after a connection has been established."

Thanks!

GamingNJncos, 2022-07-20:
Just to follow up on this I've done extensive testing with this and the library as written has a consistent 20+ms of lag with the basic examples and no additional changes. 

If you can clarify on the above request it may greatly speed up the existing latency as it is very high and should be as straightforward as at onconnect polling the connected device name (desc in example) and sending the conn_handle settings provided.

spleen1981, 2022-07-25:
#118 should do the job.
Hardcoded min/max intervals should provide low enough latency (at least it's what I'm using).

GamingNJncos, 2022-07-26:
Excellent. I'll give these changes a try and run some more tests. Thank you!

closed 2022-07-26 by GamingNJncos