lemmingDev/ESP32-BLE-Gamepad #111: is it possible to use bluetoothserial.h???

mbohben, 2022-07-02:
hello @lemmingDev ,
just straight as the issue does, im wondering on how to make BLE-Gamepad to work with simhub read message too. since im doing some f1 wheel project for my simrig. there are some sample from my code.

`
#include <Arduino.h>
#include "BluetoothSerial.h"
#include <FastLED.h>
#include <keypad.h>
#include <BleGamepad.h>

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BleGamepad bleGamepad("F1Wheel", "mbohben", 100); // Shows how you can customise the device name, manufacturer name and initial battery level

//def
#define BUF_SIZE 64
#define REV_LIGHTS_COUNT 16
#define WS2812_PIN 33
#define ROWS 5
#define COLS 4
#define numOfButtons 4

//fastled
char simHubMessageBuf[BUF_SIZE];
BluetoothSerial spp;
CRGB leds[REV_LIGHTS_COUNT];

const byte ColorNone[3] = {0, 0, 0};
const byte ColorOrange[3] = {168, 80, 0};
const byte ColorBlue[3] = {0, 0, 168};
const byte ColorGreen[3] = {0, 168, 0};
const byte ColorYellow[3] = {168, 168, 0};
const byte ColorRed[3] = {168, 0, 0};

int spd;
int revs;

//keypad
uint8_t rowPins[ROWS] = {13, 12, 14, 27, 26}; // ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {15, 4, 16, 17}; // ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
uint8_t keymap[ROWS][COLS] =
    {
      {1,2,3,4},
      {5,6,7,8},
      {9,10,11,12},
      {13,14,15,16},
      {17,18,19,20}
    };

Keypad customKeypad = Keypad(makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

//direct button
byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = {23, 22, 1 , 3};
byte physicalButtons[numOfButtons] = {21, 22, 23, 24};

void setup() 
{
    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
    bleGamepadConfig.setButtonCount(24);
    bleGamepadConfig.setHatSwitchCount(0);
    bleGamepadConfig.setWhichSpecialButtons(false, false, false, false, false, false, false, false);
    bleGamepadConfig.setWhichAxes(false, false, false, false, false, false, false, false);
    bleGamepadConfig.setWhichSimulationControls(false, false, false, false, false);
    Serial.begin(115200);
    Serial.println("setup");

  memset(simHubMessageBuf, 0x0, BUF_SIZE);
  bleGamepad.begin(&bleGamepadConfig);
  spp.begin();

  FastLED.addLeds<NEOPIXEL, WS2812_PIN>(leds, REV_LIGHTS_COUNT);

  for (byte currentPinIndex = 0; currentPinIndex < numOfButtons; currentPinIndex++)
    {
        pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
        previousButtonStates[currentPinIndex] = HIGH;
        currentButtonStates[currentPinIndex] = HIGH;
    }
}

void loop() 
{
  if (spp.available() > 0) {
    spp.readBytesUntil('{', simHubMessageBuf, BUF_SIZE);
    int readCount = spp.readBytesUntil('}', simHubMessageBuf, BUF_SIZE);
    simHubMessageBuf[min(readCount - 1, BUF_SIZE - 1)] = 0x0;
    processMessage();
    memset(simHubMessageBuf, 0x0, BUF_SIZE);
  }
  FastLED.show();
  KeypadUpdate();
  delay(10);

  if (bleGamepad.isConnected())
    {
        for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
        {
            currentButtonStates[currentIndex] = digitalRead(buttonPins[currentIndex]);

            if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex])
            {
                if (currentButtonStates[currentIndex] == LOW)
                {
                    bleGamepad.press(physicalButtons[currentIndex]);
                }
                else
                {
                    bleGamepad.release(physicalButtons[currentIndex]);
                }
            }
        }

        if (currentButtonStates != previousButtonStates)
        {
            for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
            {
                previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
            }

            bleGamepad.sendReport();
        }

        delay(20);
    }
}

void processMessage() {
  char msgType = simHubMessageBuf[0];

  switch (msgType) {
    case 'R': {
      sscanf(&simHubMessageBuf[1], "%d", &revs);
      int numOfLightsToShow = round((revs / 100.0f) * REV_LIGHTS_COUNT);
      for(int i=0; i < REV_LIGHTS_COUNT; i++) {
        const byte *color = i < numOfLightsToShow ? ledColor(i) : ColorNone;
        leds[i].setRGB(color[0], color[1], color[2]);
      }
      break;
    }
    case 'S':
      sscanf(&simHubMessageBuf[1], "%d", &spd);
      break;
  }

  Serial.print("Revs: ");
  Serial.println(revs);
  Serial.print("Speed: ");
  Serial.println(spd);
}

const byte* ledColor(int index) {
  switch(index) {
    case 16:
      return ColorBlue;
    case 15:
    case 14:
    case 13:
    case 12:
      return ColorRed;
    case 11:
    case 10:
    case 9:
    case 8:
      return ColorOrange;
    case 7:
    case 6:
    case 5:
    case 4:
      return ColorYellow;
    case 3:
    case 2:
      return ColorGreen;
    default: 
      return ColorNone;  
  }
}

void KeypadUpdate()
{
    customKeypad.getKeys();

    for (int i = 0; i < LIST_MAX; i++) // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
    {
        if (customKeypad.key[i].stateChanged) // Only find keys that have changed state.
        {
            uint8_t keystate = customKeypad.key[i].kstate;

            if (bleGamepad.isConnected())
            {
                if (keystate == PRESSED)
                {
                    bleGamepad.press(customKeypad.key[i].kchar);
                } // Press or release button based on the current state
                if (keystate == RELEASED)
                {
                    bleGamepad.release(customKeypad.key[i].kchar);
                }

                bleGamepad.sendReport(); // Send the HID report after values for all button states are updated, and at least one button state had changed
            }
        }
    }
}
`

lemmingDev, 2022-07-02:
Hi

The current version of the library uses NimBLE, which unfortunately does not allow the simultaneous use of bluetoothserial

An older version of my library will probably be compatible
There have been a few people ask. Perhaps I will create an up to date version of the library that uses the older/standard Bluetooth library that is compatible. I've got 2 weeks off in October and 5 weeks off in Dec/Jan. will attack it then, unless someone wants to beat me to it...

mbohben, 2022-07-03:
> Hi
>
> The current version of the library uses NimBLE, which unfortunately does not allow the simultaneous use of bluetoothserial
>
> An older version of my library will probably be compatible There have [...]

thanks for your fast response sir, will try to figure it out on the older version

lemmingDev, 2022-09-21:
So, got around to investing the time (turns out it was less than 1 hour) needed to get the current v5.1 version of the library ported back to the original BT stack instead of NimBLE

This should allow the use of BluetoothSerial.h, though no testing has ben done...

Let me know how you go
[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/9620632/ESP32-BLE-Gamepad.zip)

lemmingDev, 2022-09-22:
So, did some testing and didn't work by default

Sees the timing to get both working at the same time is a bit special

What I will do is make it so Classic BT is built in to my library with the begin method called where it needs to be and then you can choose to enable it and send data 

Did some quick testing along those lines and I was able to have the BLE Gamepad working at the same time as sending Classic BT Serial messages

Will need a little more development and testing, but it looks promising

lemmingDev, 2022-09-22:
Actually, given more testing, it seems to work fine as expected, except:
1. You need to pair the BL Classic device first if pairing to both on Windows
    (connects perfectly to both again on powerup)(not sure about other devices)
2. BLE and BT Classic both use the BLE name
3. Other stuff I'm not aware of yet

Here is a working example of a single button pressed that turns an LED on, presses button 1 on the controller, and also sends a message over Serial BT when the pin is activated

```
#include <Arduino.h>
#include <Bounce2.h>    // https://github.com/thomasfredericks/Bounce2
#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad
#include "BluetoothSerial.h"

#define BOUNCE_WITH_PROMPT_DETECTION // Make button state changes available immediately
#define BUTTON_PIN 39
#define LED_PIN 14

Bounce debouncer = Bounce(); // Instantiate a Bounce object
BleGamepad bleGamepad;       // Instantiate a BleGamepad object
BluetoothSerial SerialBT;    // Instantiate a BluetoothSerial object

int currentButtonState = 1;
int previousButtonState = 1;

void setup()
{
    SerialBT.begin();   // Begin the BT Classic serial port
    bleGamepad.begin(); // Begin the BLE gamepad
      

    pinMode(BUTTON_PIN, INPUT_PULLUP); // Setup the button with an internal pull-up

    debouncer.attach(BUTTON_PIN); // After setting up the button, setup the Bounce instance :
    debouncer.interval(5);        // interval in ms

    pinMode(LED_PIN, OUTPUT); // Setup the LED :
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        debouncer.update(); // Update the Bounce instance

        int currentButtonState = debouncer.read(); // Get the updated value

        // If there is a state change
        if (currentButtonState != previousButtonState)
        {
          // Press/release gamepad button and turn on or off the LED as determined by the state
          if (currentButtonState == LOW)
          {
              digitalWrite(LED_PIN, HIGH);
              SerialBT.println("Button Pressed");
              bleGamepad.press(BUTTON_1);
          }
          else
          {
              digitalWrite(LED_PIN, LOW);
              SerialBT.println("Button Released");
              bleGamepad.release(BUTTON_1);
          }
        }

        // Set previous button state based on current button state
        previousButtonState = currentButtonState;
    }
}
```

mbohben, 2022-09-22:
Depending on the code above, so the device will start as 2 different instance (bleGamepad and serialBT)?

mbohben, 2022-09-22:
Depending on the code above, so the device will start with 2 instance (bleGamepad and serialBT)?

lemmingDev, 2022-09-22:
Yeah - one of each

Here is some more code I just tested, where I connected the BT Classic to my phone.
This sketch also has serial passthrough, where all data sent from BT Classic serial is output to USB serial on Arduino and vice versa

Worked perfectly. Everything I typed on my phone was sent to the Arduino serial, and everything I typed on Arduino serial was sent to the phone --> all with the BLE gamepad still functioning fine and sending notifications to the phone on button presses

```
#include <Arduino.h>
#include <Bounce2.h>    // https://github.com/thomasfredericks/Bounce2
#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad
#include "BluetoothSerial.h"

#define BOUNCE_WITH_PROMPT_DETECTION // Make button state changes available immediately
#define BUTTON_PIN 39
#define LED_PIN 14

Bounce debouncer = Bounce(); // Instantiate a Bounce object
BleGamepad bleGamepad;       // Instantiate a BleGamepad object
BluetoothSerial SerialBT;    // Instantiate a BluetoothSerial object

int currentButtonState = 1;
int previousButtonState = 1;

void setup()
{
    SerialBT.begin();     // Begin the BT Classic serial port
    bleGamepad.begin();   // Begin the BLE gamepad
    Serial.begin(115200); // Begin USB serial port

    pinMode(BUTTON_PIN, INPUT_PULLUP); // Setup the button with an internal pull-up

    debouncer.attach(BUTTON_PIN); // After setting up the button, setup the Bounce instance :
    debouncer.interval(5);        // interval in ms

    pinMode(LED_PIN, OUTPUT); // Setup the LED :
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        if (Serial.available()) 
        {
          SerialBT.write(Serial.read());
        }
        
        if (SerialBT.available()) 
        {
          Serial.write(SerialBT.read());
        }
        
        debouncer.update(); // Update the Bounce instance

        int currentButtonState = debouncer.read(); // Get the updated value

        // If there is a state change
        if (currentButtonState != previousButtonState)
        {
          // Press/release gamepad button and turn on or off the LED as determined by the state
          if (currentButtonState == LOW)
          {
              digitalWrite(LED_PIN, HIGH);
              SerialBT.println("Button Pressed");
              bleGamepad.press(BUTTON_1);
          }
          else
          {
              digitalWrite(LED_PIN, LOW);
              SerialBT.println("Button Released");
              bleGamepad.release(BUTTON_1);
          }
        }

        // Set previous button state based on current button state
        previousButtonState = currentButtonState;
    }
}
```

lemmingDev, 2022-09-22:
I might even make this the default version of BLE stack so that everyone can use BT Classic, and have a seperate branch with NimBLE incase people want to have the benefit of reduced memory usage etc. 

There have been a number of people asking for this over the past few years.
I think people creating custom controllers for sim racing etc have been wanting it for a while

mbohben, 2022-09-22:
I'll try it as soon as my work hour done, thank you very much for the response.

lemmingDev, 2022-09-22:
Played around with it some more.

When using both BLE and BT Classic, Android only sees the BT Classic and not the BLE
Windows sees both and can connect to both, either, or on two different PCs, or BLE on PC and BT Classic on Android
Also, to get the BLE Gamepad paired on PC, you need to reset the ESP32 and connect immediately

Once it's setup, it re-pairs automatically, so I'm not really too worried about trying to track down a fix at the moment

mbohben, 2022-09-23:
done rebuilding my code, will test it as soon as possible got a race to attend too this morning

sefatunckanat, 2022-10-27:
> I might even make this the default version of BLE stack so that everyone can use BT Classic, and have a seperate branch with NimBLE incase people want to have the benefit of reduced memory usage [...]

You should check this one for sim racers. This is exactly what we want to do. 
https://dev.to/nmwilk/simhub-without-the-wires-18ij

twinking92, 2023-03-25:
> So, got around to investing the time (turns out it was less than 1 hour) needed to get the current v5.1 version of the library ported back to the original BT stack instead of NimBLE
>
> This should [...]

Hi,
I tested this Version today and it works for me with classic BT and BLE. Do you plan to make the changes also in the main branch?
Kind regards

LeeNX, 2023-10-29:
Would it not be better to extended the library to use the ```Nordic UART Service (NUS)``` (https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/libraries/bluetooth_services/services/nus.html#nordic-uart-service-nus)

Like the BLE-UART example in both nimBLE and blueDriod libraries.  
  
Would need to add the services and TX/RX Characteristic with callbacks for the incoming data.

LeeNX, 2024-11-23:
Finally got around to drawing up a sketch - https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/195#issuecomment-2488799340

This worked in my testing between ESP32-C3 connected to my Android phone and my Mac connected via serial for debugging and monitoring, where I could send and receive data between BLE connected devices.  
  
Worked pretty well and I think adding a screen where you can pop up notices or other things on the gamepad. Just not sure how or what you might be doing with the host OS.

ivomarsan, 2025-01-23:
@lemmingDev I was needing to use BluetoothSerial to get simhub data. What was the last version of your package without nimBLE?

lemmingDev, 2025-01-24:
> [@lemmingDev](https://github.com/lemmingDev) I was needing to use BluetoothSerial to get simhub data. What was the last version of your package without nimBLE?

Probably the one here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254260699

It's been tested to work using sketches such as this one https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254595287

null-except, 2025-04-11:
To have [this code working ](https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254595287) should we use an older release? I tried to run it but without success, I'm not really sure on how to do it properly.

lemmingDev, 2025-04-11:
Did you use the release here?
Probably the one here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254260699

null-except, 2025-04-11:
Yes indeed but maybe I need to review my method, I just replaced the files in my environment but maybe PlatformIO replace them again.

null-except, 2025-04-11:
Ok so before I was trying to use: 

`lib_deps = lemmingdev/ESP32-BLE-Gamepad@0.5.1`

But it was not compiling, after putting [this release](https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254260699) directly into the lib folder it kind of work!

 There's a lot of errors in the USB Serial monitor, but I'm able to connect to the port COM9 and send a message! 

[image]

Here is my code, maybe I'm missing something to have this much of errors? 

```
#include <Arduino.h>
#include <BleGamepad.h>

#include "BluetoothSerial.h"
BluetoothSerial SerialBT;

const int ledPins[9] = {13, 12, 14, 25, 26, 27, 23, 16, 17};

const int analogHallPin = 15;
const int hallButton1 = 32;
const int hallButton2 = 33;

const int axisMin = 896;
const int axisMax = 1890;

BleGamepad bleGamepad("Orange And Black Wheel", "Adrien", 70);

// Matrice de boutons
const byte ROWS = 3;
const byte COLS = 4;
byte rowPins[ROWS] = {2, 0, 5};
byte colPins[COLS] = {18, 19, 21, 22};
bool buttonStates[ROWS][COLS] = {false};
int baseButtonIndex = 3;

void initLed()
{
    for (int i = 0; i < 9; i++)
    {
        pinMode(ledPins[i], OUTPUT);
        digitalWrite(ledPins[i], LOW);
    }

    for (int i = 0; i < 9; i++)
    {
        delay(100);
        digitalWrite(ledPins[i], HIGH);
    }

    delay(250);

    for (int i = 0; i < 9; i++)
    {
        delay(100);
        digitalWrite(ledPins[i], LOW);
    }
}

void setup()
{
    Serial.begin(115200);
    analogReadResolution(12);

    pinMode(hallButton1, INPUT_PULLUP);
    pinMode(hallButton2, INPUT_PULLUP);

    for (int i = 0; i < ROWS; i++)
    {
        pinMode(rowPins[i], OUTPUT);
        digitalWrite(rowPins[i], HIGH);
    }

    for (int i = 0; i < COLS; i++)
    {
        pinMode(colPins[i], INPUT_PULLUP);
    }

    SerialBT.begin();
    delay(1000);
    Serial.println("SerialBT ok");

    initLed();

    bleGamepad.begin();
    delay(1000);
    Serial.println("bleGamepad ok");
}

void loop()
{

    int hallValue = analogRead(analogHallPin);
    hallValue = constrain(hallValue, axisMin, axisMax);
    int16_t xAxis = map(hallValue, axisMin, axisMax, 0, 32767);

    // Serial.println("hallValue: " + String(hallValue) + " | xAxis: " + String(xAxis));

    if (bleGamepad.isConnected())
    {
        bleGamepad.setX(xAxis);

        bool btn1 = digitalRead(hallButton1) == HIGH;
        bool btn2 = digitalRead(hallButton2) == HIGH;

        if (btn1)
            bleGamepad.press(BUTTON_1);
        else
            bleGamepad.release(BUTTON_1);

        if (btn2)
            bleGamepad.press(BUTTON_2);
        else
            bleGamepad.release(BUTTON_2);

        for (int row = 0; row < ROWS; row++)
        {
            digitalWrite(rowPins[row], LOW);
            delayMicroseconds(5);

            for (int col = 0; col < COLS; col++)
            {
                bool pressed = digitalRead(colPins[col]) == LOW;
                int buttonIndex = baseButtonIndex + (row * COLS + col);

                if (pressed && !buttonStates[row][col])
                {
                    bleGamepad.press(buttonIndex);
                    buttonStates[row][col] = true;
                }
                else if (!pressed && buttonStates[row][col])
                {
                    bleGamepad.release(buttonIndex);
                    buttonStates[row][col] = false;
                }
            }

            digitalWrite(rowPins[row], HIGH);
        }

        bleGamepad.sendReport();
    }

    if (SerialBT.available())
    {
        Serial.write(SerialBT.read());
    }

    delay(4);
}
```

Can the `[172760][E][BLECharacteristic.cpp:537] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error` be related to my hardware? It's seems it only appear when I open the Arduino Serial Monitor windows... 

I'm using an ESP32-WROOM from [Freenove](https://store.freenove.com/products/fnk0090).

lemmingDev, 2025-04-11:
Yeah, that version I linked is an unofficial port of 0.5.1, and needs to be copied directly

I just read your edit and see that it's now working with a loop delay of 10

Great to hear. What are you going to use the classic serial data for?

null-except, 2025-04-11:
Same as other people, I'm sending a Serial stream from SimHub to my wheel to light up some leds :)

lemmingDev, 2025-04-11:
Is RP Pico W more common now, or are people still using ESP32 for SimHub?

I've never really looked into it

null-except, 2025-04-12:
Interesting, I rely a lot on the esp32 chip for my DIY projects, I must be pretty out of date with my chip selection I guess ha!