lemmingDev/ESP32-BLE-Gamepad #112: Can't identify default L and R buttons to send the command

AgustinCingolani, 2022-07-06:
I'm building a SNES Bluetooth controller in order to connect to Blueretro.

This is all working except L and R buttons. I've identified the phisical signal but I can't send the corresponding command (Like I've with the PS4 controller). I tried one by one from 1 to 50.

Is there a special command like the setHat1 for the D-PAD?

Here is the button mapping:

D-PAD -> Hat1
START -> BUTTON 12
SELECT -> BUTTON 11
A -> BUTTON 2
B -> BUTTON 1
X -> BUTTON 5
Y -> BUTTON 4

Thanks in advance

lemmingDev, 2022-07-26:
It's OS dependant, eg Windows generally sees buttons 5 and 6 as L and R

GamingNJncos, 2022-07-26:
Just to chime in here - make sure you are defining your "trigger" as a button here.  If you try to use the axis based setting it will cause issues

For example  this wont work immediately for digital use:
```
 bleGamepad.setLeftTrigger(32767);       // or bleGamepad.setRX(32767);
 bleGamepad.setRightTrigger(32767);      // or bleGamepad.setRY(32767);
```

Instead try
```
 bleGamepad.press(BUTTON_5);
 bleGamepad.release(BUTTON_5); 
```

lemmingDev, 2022-08-20:
No activity  - so closing

closed 2022-08-20 by lemmingDev