lemmingDev/ESP32-BLE-Gamepad #114: ESP32-BLE-Gamepad sending/update battery level

APiRacing, 2022-07-11:
Guys, I need help, I'm new to this business, help me implement in my code, a part of the code for reading the battery level, so that win10 displays the level in BT devices, let it work even if you need to reconnect the device to get up-to-date information about battery level

[WirelessGameHub.txt](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/9083629/WirelessGameHub.txt)

lemmingDev, 2022-07-12:
Hi

From my testing, the battery level can be correctly set on connection, but then doesn't update while in use.

Not sure if this is a library limitation, or an operating system limitation.

As for reading the battery level of an ESP32 device, there are multiple tutorials out there

APiRacing, 2022-07-13:
> Hi
> 
> From my testing, the battery level can be correctly set on connection, but then doesn't update while in use.
> 
> Not sure if this is a library limitation, or an operating system limitation.
> 
> As for reading the battery level of an ESP32 device, there are multiple tutorials out there

Hello friend, please help me with your tutorial, I don’t understand almost anything in programming, but the way it works for you completely suits me

APiRacing, 2022-07-15:
> Hi
> 
> From my testing, the battery level can be correctly set on connection, but then doesn't update while in use.
> 
> Not sure if this is a library limitation, or an operating system limitation.
> 
> As for reading the battery level of an ESP32 device, there are multiple tutorials out there

Hi, where are you from?
can we contact in any messenger personally?

spleen1981, 2022-07-24:
> Not sure if this is a library limitation, or an operating system limitaSettion.

`setBatteryLevel` in NimBLE was just setting the value, without any notification.
Exposed now the relevant characteristic [upstream](https://github.com/h2zero/NimBLE-Arduino/pull/422) and PR #117 submitted here.

APiRacing, 2022-07-25:
> > Not sure if this is a library limitation, or an operating system limitaSettion.
> 
> `setBatteryLevel` in NimBLE was just setting the value, without any notification. Exposed now the relevant characteristic [upstream](https://github.com/h2zero/NimBLE-Arduino/pull/422) and PR #117 submitted here.

void BleGamepad::setBatteryLevel(uint8_t level)
{
    this->batteryLevel = level;
    if (hid != 0)
    if (hid != 0){
        this->hid->setBatteryLevel(this->batteryLevel);
	this->hid->batteryLevel()->notify();
    }
}

void BleGamepad::taskServer(void *pvParameter)
I need to insert these lines into my code?

spleen1981, 2022-07-25:
> I need to insert these lines into my code?

No, code in user sketch is same as before, modification applies to the ESP32-BLE-Gamepad library.
Once it's merged, you just compile your sketch with updated library (and updated NimBLE library upstream).
Or you can apply the [patch](https://github.com/lemmingDev/ESP32-BLE-Gamepad/commit/5fa76aeef6542d6da6e17119a379df107b51855b.patch) by yourself locally.

APiRacing, 2022-07-25:
> No, code in user sketch is same as before, modification applies to the ESP32-BLE-Gamepad library. Once it's merged, you just compile your sketch with updated library (and updated NimBLE library [...]

write please what exactly I need to do, I do not understand anything((

APiRacing, 2022-07-25:
> No, code in user sketch is same as before, modification applies to the ESP32-BLE-Gamepad library. Once it's merged, you just compile your sketch with updated library (and updated NimBLE library [...]

i want to make a gaming device on lolin 32 board
I have a code with firmware, it works, but I need to display the charge level in the windows system, even if it is through constant reconnection, but this is already much better than nothing

APiRacing, 2022-07-25:
> No, code in user sketch is same as before, modification applies to the ESP32-BLE-Gamepad library. Once it's merged, you just compile your sketch with updated library (and updated NimBLE library [...]

Send me your email, I would like to contact you personally

spleen1981, 2022-07-25:
> write please what exactly I need to do, I do not understand anything((

Basically you don't need to do anything, just wait that gamepad library and nimble library new versions are released and compile your current sketch.

APiRacing, 2022-07-25:
> > write please what exactly I need to do, I do not understand anything((
> 
> Basically you don't need to do anything, just wait that gamepad library and nimble library new versions are released and compile your current sketch.

I have now downloaded the NimBLE library
Will I need to include it in the sketch itself?

APiRacing, 2022-07-25:
> > write please what exactly I need to do, I do not understand anything((
> 
> Basically you don't need to do anything, just wait that gamepad library and nimble library new versions are released and compile your current sketch.

and when do we expect the updated BLE gamepad library?
Or is it already possible to download somewhere to test it?

APiRacing, 2022-07-25:
> > write please what exactly I need to do, I do not understand anything((
> 
> Basically you don't need to do anything, just wait that gamepad library and nimble library new versions are released and compile your current sketch.

I'm just wondering how it works if nothing is written in the code about updating the charge level

closed 2022-07-25 by lemmingDev

lemmingDev, 2022-08-20:
// Set battery level to 87% <-- report must be sent after
bleGamepad.setBatteryLevel(87);

alkurop, 2024-04-01:
Hi. The battery level notification is not working for me on Anrdoid.
I checked the fix code - I think it should be ` this->hid->batteryLevel()->notify(true);`, because `notify` function has a parameter.

PR #117 (2022-07-25): fix battery level not updating
Needs https://github.com/h2zero/NimBLE-Arduino/commit/f699ab8395c3caf895c5d2a8ff82bf48e3442d32 at least.
Fixes #114 and #85