lemmingDev/ESP32-BLE-Gamepad #116: Right Thumbstick Y Axis

lewisjustin749, 2022-07-15:
Trying to configure the right joystick and I'm noticing that Rz is controlling the left thumbstick's X-axis. What is the proper way to set up a right thumbstick so that it's properly recognized?

lemmingDev, 2022-07-26:
Hi

I've just checked the library and it's working as intended

In the HID report that is used to set the gamepad up, there is no Left Thumbstick or Right Thumbstick defined

It's just the operating system that usually sees X and X axes as the left thumbstick and Z and rZ axes as the right thumbstick (and rX and rY as the triggers)

My guess is that you're on an OS other than Windows yeah?

lemmingDev, 2022-07-29:
Let me know if you need me to re-open this

closed 2022-07-29 by lemmingDev