lemmingDev/ESP32-BLE-Gamepad #120: NimBLEHIDDevice' has no member named 'batteryLevel'

marcelomanzo, 2022-07-26:
My CI process started failing this morning, looks like NimBLE changed or removed batteryLevel method.

$ arduino-cli --log-level error compile -e --fqbn esp32:esp32:esp32 arduino/magicwheel/
/root/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepad.cpp: In member function 'void BleGamepad::setBatteryLevel(uint8_t)':
/root/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepad.cpp:1297:17: error: 'class NimBLEHIDDevice' has no member named 'batteryLevel'
      this->hid->batteryLevel()->notify();
                 ^
Used library      Version Path                                                             
ESP32-BLE-Gamepad 0.4.1   /root/Arduino/libraries/ESP32-BLE-Gamepad                        
NimBLE-Arduino    1.3.8   /root/Arduino/libraries/NimBLE-Arduino

marcelomanzo, 2022-07-26:
More info here:
https://github.com/h2zero/NimBLE-Arduino/commit/f699ab8395c3caf895c5d2a8ff82bf48e3442d32

lemmingDev, 2022-07-26:
Setting the battery level works perfectly
Until NimBLE Arduino push a new update however, you'll need to download NimBLE Arduino from source and replace the one in your libraries folder as their version 1.4.0 was released 16 days ago, and the battery fix was done 2 days ago

Fabi0San, 2022-07-27:
Alternatively, one can stick to ESP32-BLE-Gamepad release 4.1

closed 2022-07-28 by lemmingDev