lemmingDev/ESP32-BLE-Gamepad #121: Axes swing from near min to near max with values -1, +1, on some applications.

Fabi0San, 2022-07-29:
I used multiple online gamepad testers and all report this behavior.
Moving from 0(center) to 1 sends it to extreme position 99.998% of range, increasing  value to say 1000 continues to increase it slightly to 99.999% of the range. 
Max value of 32k gets it to 100% as expected. 
These testers work fine with an xbox controller.

The gamepad tester in the device properties page in windows work as expected and show gradual increase from -32k to +32k.

Are we missing something about the protocol and that bad behavior may surface in some games as well?

https://gamepad-tester.com/
https://gamepadviewer.com/

lemmingDev, 2022-07-30:
Hi

I'm no expert in this field by any means, but I can say that the HID descriptor for this library clearly states the min and max values correctly and I feel it's those websites that are dealing with 1 and -1 in a funny way, perhaps aiming to be compatible with a larger range of devices. 

I think some OS may have trouble with a resolution so large and I have contemplated making the axes resolution selectable.

Anyway, in the mean time, you can code it so that those values close to zero are never sent...

Fabi0San, 2022-07-31:
I think you misunderstood, the problem is not with 1 and -1, 2 and -2 are also in the wrong place, the whole scale is all messed up.  The only numbers that are right are 0, 32k and -32k.

lemmingDev, 2022-08-01:
Hi

I see what you mean now
So, the issue is that those testing programs don't like negative values

Try this edited BleGamepad.cpp file

0 for axis min
16,384 for axis center
32767 for axis max

Make sure you remove your old Bluetooth device from the Bluetooth control panel and then add the new one

I have tested on https://gamepad-tester.com/ for both 32767 and 255 max and they're working fine, whereas before, I was getting the same outcome as you

NB. If you search for 255 in that file, you can see how to make 255 the max instead of 32767 if you wish

Please report back

Next version of the library, I will allow some type of selection for these values

Fabi0San, 2022-08-01:
oh I see, they interpret those as unsigned and get confused when max<min. I wonder if that is that app doing it wrong or the HTML5 routine they all use. It could also be that they all copied buggy javascritpt from one another. 

Either way it would be nice to expose min and max on our configuration class, keeping the defaults at -32k +32k for backwards compat. LMK if you want me to work on that.

Thanks for looking!

lemmingDev, 2022-08-19:
Feel free to make a pull request if you like as that would be quicker for everyone, or I will get around to it eventually.

Thanks for your interest in the library

lemmingDev, 2022-08-20:
Actually - found some time this morning to get it done. The library now has configurable min/max values for axes and simulation controls

Need to write an example or two

I will make the default min 0 from the next version so the library is more compatible with operating systems other than Windows

I will bump it to version 5.0 and put up a Breaking Changes notice with it

lemmingDev, 2022-08-20:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.5.0

closed 2022-08-20 by lemmingDev