lemmingDev/ESP32-BLE-Gamepad #122: compile error ESP32 DEVKIT V1

felichivas, 2022-07-30:
here what is shown when i tried to compile on ESP32 DEVKIT V1.

C:\Users\aguil\Documents\Arduino\libraries\ESP32-BLE-Gamepad-master\BleGamepad.cpp: In member function 'void BleGamepad::setBatteryLevel(uint8_t)':
C:\Users\aguil\Documents\Arduino\libraries\ESP32-BLE-Gamepad-master\BleGamepad.cpp:1297:17: error: 'class NimBLEHIDDevice' has no member named 'batteryLevel'; did you mean 'setBatteryLevel'?
      this->hid->batteryLevel()->notify();
                 ^~~~~~~~~~~~
                 setBatteryLevel
exit status 1
Error  DOIT ESP32 DEVKIT V1.

lemmingDev, 2022-07-30:
If you want to use the current version of this library, you'll either need to wait until NimBLE releases an update, or, to get it working straight away, download the current code from their GitHub and manually install it --> works fine.

Alternatively, use the previous of the gamepad library until NimBLE release an official update

felichivas, 2022-07-30:
thanks @lemmingDev, now it works

GamingNJncos, 2022-08-01:
For anyone else that runs into this in the interim just remove these 2 // from line 58 in NimBLE-Arduino/src/NimBLEHIDDevice.h
[image]

mhaebler, 2022-08-18:
Looks like https://github.com/h2zero/NimBLE-Arduino/pull/390 broke the "just download the master .zip" fix here as outlined in https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/122#issuecomment-1200113884

Filed #124 for that.

lemmingDev, 2022-08-19:
Thanks

Fixed in https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.4.4 and use NimBLE-Arduino from https://github.com/h2zero/NimBLE-Arduino/tree/master

closed 2022-08-20 by lemmingDev

jamel-86, 2023-09-08:
I'm trying to compline with ESP32-BLE-Gamepad@0.4.4 and NimBLE-Arduino@2.0.0 (followed the links you shared) 
and still get the similar error:

error: 'class BleGamepad' has no member named 'notify'
     bleGamepad.notify();
                ^~~~~~

lemmingDev, 2023-09-09:
NimBLE-Arduino 1.4.1 is the latest official version
May I suggesting using that version