lemmingDev/ESP32-BLE-Gamepad #123: how to use the triggers?

davidxgames, 2022-08-10:
Could you help me, I'm using this code, joysticks work fine, but triggers don't respond.

`// TRIGGER BUTTONS
#define LT 32
#define RT 33

// LEFT-JOYSTICK
#define LH 34
#define LV 35

// RIGHT-JOYSTICK
#define RH 39
#define RV 36

void loop()
    int value1 = map(analogRead(LH), 0, 4095, 32737, -32737);
    int value2 = map(analogRead(LV), 0, 4095, 32737, -32737);
    bleGamepad.setLeftThumb(value1, value2);

    
    int value3 = map(analogRead(RH), 0, 4095, 32737, -32737);
    int value4 = map(analogRead(RV), 0, 4095, 32737, -32737);
    bleGamepad.setRightThumb(value3, value4);

    int value5 = map(analogRead(LT), 0, 4095, 32737, -32737);
    bleGamepad.setLeftTrigger(value5);

    int value6 = map(analogRead(RT), 0, 4095, 32737, -32737);
    bleGamepad.setRightTrigger(value6);
    

`

GamingNJncos, 2022-08-10:
What are you using to test? This should work on windows.  Depending on your connected device it may prefer negative -> positive which you can test easily swapping the numbers in your map.

davidxgames, 2022-08-10:
> ¿Qué estás usando para probar? Esto debería funcionar en Windows. Dependiendo de su dispositivo conectado, puede preferir negativo -> positivo, que puede probar fácilmente intercambiando los números en su mapa.

I am testing it on android with an app to test controls, it is called gamepad tester, with this application I tested the snes control that I made and also one for nintendo 64.

lemmingDev, 2022-08-20:
Hi

The new v5.0 of this library uses 0 as the default min axis value and is more compatible

https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.5.0

Try it out

closed 2022-08-20 by lemmingDev