lemmingDev/ESP32-BLE-Gamepad #124: Compilation error due to removal of NimBLESecurity class

mhaebler, 2022-08-18:
Breaking change in upstream NimBLE-Arduino is resulting in a number of compilation errors - see: https://github.com/h2zero/NimBLE-Arduino/pull/390

```
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp: In static member function 'static void BleGamepad::taskServer(void*)':
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1318:5: error: 'NimBLESecurity' was not declared in this scope
     NimBLESecurity *pSecurity = new NimBLESecurity();
     ^~~~~~~~~~~~~~
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1318:5: note: suggested alternative: 'BLESecurity'
     NimBLESecurity *pSecurity = new NimBLESecurity();
     ^~~~~~~~~~~~~~
     BLESecurity
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1318:21: error: 'pSecurity' was not declared in this scope
     NimBLESecurity *pSecurity = new NimBLESecurity();
                     ^~~~~~~~~
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1318:21: note: suggested alternative: 'BLESecurity'
     NimBLESecurity *pSecurity = new NimBLESecurity();
                     ^~~~~~~~~
                     BLESecurity
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1318:37: error: expected type-specifier before 'NimBLESecurity'
     NimBLESecurity *pSecurity = new NimBLESecurity();
                                     ^~~~~~~~~~~~~~
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1320:38: error: 'ESP_LE_AUTH_BOND' was not declared in this scope
     pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
                                      ^~~~~~~~~~~~~~~~
C:\Users\foobar\Documents\Arduino\libraries\ESP32-BLE-Gamepad-x0.4.3\BleGamepad.cpp:1320:38: note: suggested alternative: 'ESP_EARLY_LOGD'
     pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
                                      ^~~~~~~~~~~~~~~~
                                      ESP_EARLY_LOGD
exit status 1
Error compiling for board ESP32 Dev Module.

```

mhaebler, 2022-08-19:
The fix is to replace the `NimBLESecurity` class instantiation and authentication method setting in BLEGamepad.cpp:1318-1320 with an equivalent call to 
`NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_BOND);`

lemmingDev, 2022-08-19:
Thanks!

https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.4.4

lemmingDev, 2022-08-20:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.5.0

New release has a note to download the new NimBLE-Arduino release from Github, plus adds configurable axes resolution

closed 2022-08-20 by lemmingDev