lemmingDev/ESP32-BLE-Gamepad #125: can you add rumble?

davidxgames, 2022-08-20:
can you add rumble? Is this possible to do?

lemmingDev, 2022-08-20:
I don't have the capability

@narsku said he was working on it, though hasn't mentioned any progress

https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/99

davidxgames, 2022-08-20:
> I don't have the capability
> 
> @Narsku said he was working on it, though hasn't mentioned any progress
> 
> #99

First of all, thanks for answering, and the truth is, your work is very good, and thanks to that, I have recycled my nes, snes, n64 and ps1 controls, and I will use ps3, ps4, xbox, xbox 360 and one, which I don't know anymore. they can repair, but thanks to you they will have a new opportunity.

And let's hope that @Narsku manages to find the solution.

lemmingDev, 2022-08-20:
Thanks. 

Great to see it go to good use. Coming up to 3 years since the first version.

closed 2022-08-20 by lemmingDev