lemmingDev/ESP32-BLE-Gamepad #126: Compile issue

stezo, 2022-08-21:
Hey

Thanks for putting so much effort into this. Looks very promising.

I'm currently trying to get your library working in my Arduino IDE on Linux for an ESP 32.
I'm using the latest version 5.0 of your library as well as the latest version of NIMBle.

Whenever I compile, I get the following error:

**/home/stezo/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepad.cpp: In member function 'void BleGamepad::setBatteryLevel(uint8_t)':
/home/stezo/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepad.cpp:1297:17: error: 'class NimBLEHIDDevice' has no member named 'batteryLevel'
      this->hid->batteryLevel()->notify();_**

This is for the "SingleButton" example, but happens for other templates identically.

I already tried slightly older version ob both libraries, but the error stays the same. Also tried installing from library manager as well as from Zip files. 

Please let me know what the problem could be. I'm a noob in this area, so please bare with me if I made just a stupid mistake.

Thanks.

lemmingDev, 2022-08-21:
Hi - It's on the main GitHub page (will paste the full text below) plus the v5.0 release notes - but in short, you need to use the GitHub master version of NimBLE-Arduino at https://github.com/h2zero/NimBLE-Arduino/tree/master which is 2.0, instead of the latest released version 1.4, until they release the update

**POSSIBLE BREAKING CHANGES - PLEASE READ**
A large code rebase (configuration class) along with some extra features (start, select, menu, home, back, volume up, volume down and volume mute buttons) has been committed thanks to @dexterdy

Since version 5 of this library, the axes and simulation controls have configurable min and max values The decision was made to set defaults to 0 for minimum and 32767 for maximum (previously -32767 to 32767) This was due to the fact that non-Windows operating systems and some online web-based game controller testers didn't play well with negative numbers. Existing sketches should take note, and see the DrivingControllerTest example for how to set back to -32767 if wanted

This version of the library has been tested against NimBLE-Arduino version 2 (as of publishing, 1.4 is the latest released version, so currently only available at https://github.com/h2zero/NimBLE-Arduino/tree/master)

Please see updated examples

stezo, 2022-08-22:
Thanks. now it's working. Sorry, could have seen that.

closed 2022-08-22 by stezo

lemmingDev, 2022-08-22:
All good
What will you be using the library for?