lemmingDev/ESP32-BLE-Gamepad #127: Controler looks freezed - using exemples

Couby, 2022-09-02:
Hi,

I'm working on an home-made custom flight stick, and this ESP32-BLE-Gamepad library looks exactly what I need in order to give life to this stick through ESP32.
This is a 23 buttons + 2 axis stick.

========================

Yet for the moment, I'm just trying to get **ESP32-BLE-Gamepad** working, using **examples** 😊 .

I'm a bit frustrated cos I don't have any example working 😢 .
Looks to be close, but I'm surely missing something 😓 .

========================

**Board I use** : this is NodeMCU ESP32 board I bought [here](https://www.amazon.fr/AZ-Delivery-NodeMCU-d%C3%A9veloppement-d%C3%A9nergie-successeur/dp/B074RGW2VQ/ref=sr_1_1_sspa?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=NS5IPVIZC9P&keywords=esp32&qid=1661703199&sprefix=esp32%2Caps%2C94&sr=8-1-spons&smid=A1X7QLRQH87QA3&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFSR1RWN0tHTklNVUgmZW5jcnlwdGVkSWQ9QTAzNDQ1MzkxVU01MTVEVEhWN1JTJmVuY3J5cHRlZEFkSWQ9QTA4NTYyNDFLSVhUMzM0U0pGNjgmd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl&th=1)

Using the ESP32 GetChipID sketch, I got this : ESP32-D0WDQ6 Rev 1.

I have multiple boards and tested them with some other basic sketches, all working well.

========================

The problem I have using all ESP32-BLE-Gamepad examples is the controler looks "freezed".
- Each time I've got the right data in bluetooth and controler name in game controler board or JoystickTest app.
- Using JoystickTest app, gives each time the right number of buttons and axes accordingly to the sketch.
- **but** no life, no button pressed or axis moves for examples including such actions in the loop
- no life either with MultipleButtons sketch, wiring a simple button to pins from example (using just one button)

I searched in other issues and didn't find something similar.

========================

Regarding **libraries**, I use : 
- nimBLE from github master repos has seen in #126 (I also had compilation problem at the begining but solved following the v2 orientation given)
- [arduino-esp32](https://github.com/espressif/arduino-esp32) v2.0.4 via Arduino library management panel
- ESP32-BLE-Gamepad - I tried latest release and latest github main, no change

========================

I precise I paid attention to unplug/replug, unpair/repair the ESP32 after each sketch upload.

Another detail : looking at serial monitor, there's nothing displayed, no data from serial monitor while many messages should come from the sketch execution.

========================

Any orientation or things to test is welcome 😊 .

Very best regards.

lemmingDev, 2022-09-06:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.5.1

So, use the new v5.1 of this library, plus the official v1.4.0 version of NimBLE (from library manager)

They keep updating the GitHub master version without providing an official release and it's causing breaking changes, so just use the above versions for now

closed 2022-09-06 by lemmingDev

Couby, 2022-09-07:
Hope I can close myself after this message :blush: , just wanted to confirm it is now working fine with this combination of libraries :tada: .

Many thanks !