lemmingDev/ESP32-BLE-Gamepad #130: Sync issue

iAmCodiSmith, 2022-09-11:
Cannot pair with PC or phone

Downloaded the latest release (5.1) and using NimBLE 1.4 from the library manager.
Used the SingleButton example.
Builds and flashes fine. (All the examples do)
ESP32 BLE Gamepad shows in Windows and on Android but immediately disappears when trying to connect.

Tried on 3 PCs and 1 Android. Shows on all devices, just wont connect.

lemmingDev, 2022-09-11:
Hi

Thanks for the report.

This combination has been verified to work on other people's ESP32

Have you only tried with the one board? 

When other people have had similar problems, it turns out to be the board or the cheap USB cable they've been using....

Let me know how you go

lemmingDev, 2022-09-13:
Did you end up getting it to work

Will close the issue if I don't hear back...

iAmCodiSmith, 2022-09-14:
My mistake, ordered new ones by espressif, should be here in a few days.
I had been using an amazon 3 pack, they are likely the issue.

iAmCodiSmith, 2022-09-16:
Espressif boards arrived today. Worked immediately. My fault, sorry for the trouble.

closed 2022-09-16 by iAmCodiSmith

lemmingDev, 2022-09-16:
Hi

I'm glad to hear that it worked out

I hope the library comes in handy

What do you plan on using it for?