lemmingDev/ESP32-BLE-Gamepad #131: ESP32-C3 Hangs on bleGamepad.begin

LeeNX, 2022-09-16:
Just got an ESP32-C3 (RISC-V), more looking for some feedback, not critical. Maybe I can assist by testing.

Tested some of the examples in the lib, just to make sure it was not me. Then added some serial logging, thou I am not sure I have got the serial debugging working.

Added further testing, but blinking the onboard LED before bleGamepad.begin and blinking the LED during the main loop. The serial reports and the LED blinks until bleGamepad.begin.

Also tested on an older ESP32 device, which worked as expected. Serial and LED blink during operations as expected.

I did test the ESP32-BLE examples, and bluetooth seems to work for fine.

LeeNX, 2022-09-16:
Bumped up the logging to debug, first line is just a serial, then mills and I believe the NemBLEDevice report is the bleGamepad.begin 
```
08:53:56.511 -> Starting Game work!
08:53:56.511 -> 1629
08:53:56.545 -> I NimBLEDevice: BLE Host Task Started
```

moschotto, 2022-09-17:
Encountered the same issue. Seem to be related if more than one device was paired

I mentioned it under limitations - wasn’t able to fix this issue. 

https://github.com/moschotto/OpenTX_EdgeTX_Bluetooth_GamePad

Fails here (see below) when 2 devices are paired (after 2nd devices was paired and then power-cycled)

https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/src/nimble/nimble/controller/src/ble_ll.c

ble_npl_event_run(ev);

lemmingDev, 2022-09-18:
So, it's working apart from this problem (which seems to have a know workaround)?

I don't have one of those boards

LeeNX, 2022-09-18:
@lemmingDev do you have a Patreon link or something that we could support and help possible get an ESP32-C3? My current choice of dev tool is https://wiki.dfrobot.com/SKU_DFR0868_Beetle_ESP32_C3. 
  
Added ESP32-C3-0.42LCD too my next wish list order.  
  
Thanks @moschotto for the hint. I will investigate. Feels like a bug, was there an issue opened for this bug?  
  
I did think I had ESP32-BLE-Gamepad instance running and working, but did also think I possible imagine that. I will try reflashing, thou I did try standard BLE sketch (BLE_write) and that worked fine.

I will update as I do further testing.

moschotto, 2022-09-18:
BTW i did not encounter this issue with a D1 mini ESP32