lemmingDev/ESP32-BLE-Gamepad #132: About Pins limitation

Couby, 2022-09-18:
Hi,

I'm facing a problem I think due to ESP32 pins limitation.
Would be cool to have some expertise on this 😊 .

I'm (trying) making a joystick with 2 analog axis + 23 buttons (without multiplexing management, just direct wiring).
I thought I had the right number of digital input available, accordingly to the table I found here : https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Yet it doesn't work 😢 .

I have my two analog entries wired on GPIO 36 and 39. Ok ✔️ 

I have 19 digital inputs wired on GPIOs 2, 26, 25, 14, 33, 27, 19, 4, 32, 13, 15, 12, 17, 16, 23, 3, 21, 22, 18. Ok ✔️ 

For the 4 remaining inputs wired on GPIOs 0, 5, 34, 35, in doesn't work. NOk ❌

I checked all wirings/buttons, they are all good and working.

A table to sum up this project :
[image]

Any ideas were I am making a mistake ?

lemmingDev, 2022-09-18:
Pins 34 and 35 don't have pullup resistors on them (nor do 36 and 39, but you're using them for analog anyway so it shouldn't matter), so you might have to add your own to get them to work correctly.

Not too sure why 0 and 5 are not working. I've got a board that uses pin 0 as an onboard button and it works fine.

LeeNX, 2022-09-18:
Dig the idea, very nice to avoid any latency introduced by multiplexers. Something I had planned to get around to play with.  
  
Not an ESP32 pin expert, but I like to referrer to https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32

For GPIO 34 & 35, you not trying to use an internal pullup or pulldown? Seems like you can't do that for them pins.  
  
GPIO 0 might have some boot up restriction, that might be work avoiding.

LeeNX, 2022-09-18:
Hahaha @lemmingDev beat me by 24seconds ... Hope that is helpful.

Couby, 2022-09-18:
Many thanks both for your quick feedback. 👍 
Effectively I didn't add pullup resistors for 34 and 35.  😅 
I'll test this to confirm. If I can get 2 more inputs for my immediate project would be perfect, i can reduce my entries to 21 buttons.

I have another type of ESP32 board with a bit different mapping. But I'll look into this later. First I'll dig this topic with this one.

Couby, 2022-09-18:
I confirm ❗ 
With pullup resistors GPIO 34 and 35 are working as digital input. 😊 

GPIO 5 is working fine now. Didn't change anything but it works. 😵 

For GPIO 0 I think effectively boot up restriction may limit it. Possibly it depends on the board ?

Looks almost resolved. I close my issue. If someone see a way to make clearer how to use GPIO 0 as input without boot interference, it will always be welcome (input def after boot up delay ?...).

Many thanks @lemmingDev and @LeeNX for your answers.

closed 2022-09-18 by Couby

lemmingDev, 2022-09-18:
Nice 

You could try adding a resistor to pin 0 as well, like you did for the other two