lemmingDev/ESP32-BLE-Gamepad #133: Connected but not working on Android Devices

satriamahen, 2022-09-21:
I'm using the TestAll example code, it works on Windows flawlessly.
But when it is connected to Android, it is detected but no input at all.

lemmingDev, 2022-09-21:
Hi

Thanks for making contact

I just tried on 2 Android phones and it worked fine for me

Maybe Windows was trying to still connect to it

Try setting the device name to something different:

Instead of
'''BleGamepad bleGamepad;

try

'''BleGamepad bleGamepad("TEST");

satriamahen, 2022-09-21:
Thanks for the response

I tried changing the device name to something different and still got the same problem

But, after I downgraded the BLE Gamepad Library to the older version that not using the NimBLE Library, it works now.
I don't know exactly why, but I think the problem is with the NimBLE Library. At least for me.

closed 2022-09-21 by satriamahen

lemmingDev, 2022-09-21:
Hi

Can you try out the attached version of the library?
It's the current version of the library, except using the orginal BT stack instead of NimBLE
[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/9620489/ESP32-BLE-Gamepad.zip)

lemmingDev, 2022-09-25:
Did you get a chance to try it out?