lemmingDev/ESP32-BLE-Gamepad #135: Meta Quest2 Bluetooth connection issue

remsurge, 2022-09-27:
I'm developing a custom controller for the Quest2d (controller Sparkfun ESP32-thing plus). When testing the controller with an android phone it works fine, pairs and stays paired, thru power cycles on both phone and controller. But when using with the quest2 the controller pairs correctly and works in the game. But if you power cycle ether device the quest2 shows that its connected to the controller but the control signals don't come thru? I've tried a lot of things, but currently my only work around is unpairing the devices before powering down and new pairing each time the quest2 is turned on. 
I thought maybe its the auto report hanging it up but changed that with same outcome. I think its something in the quest2 programing but wondering if anyone knows of a way to make the devices forget/redue their original key/handshake each time their powered on?
Or any other thoughts on why this is happening! 
I'm using a stripped down version of one of the example codes provided. 

thanks for the work on this!!

lemmingDev, 2022-09-30:
Can you try this version
[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/9687407/ESP32-BLE-Gamepad.zip)
It's basically v5.1 of my library that uses the default BT stack instead of the NimBLE stack
Let me know if it works better...

lemmingDev, 2022-10-09:
Did you get a chance to try this?

Kusachev, 2022-10-11:
We had very similar issues. ESP32 device with the ESP32-BLE-Gamepad lib with NimBLE doesn't reconnect automatically to Android 11 and Android 9 based devices (but works well with Android 12 based device). The library you provided here works well with all of them.
Steps to reproduce:
1. Pair ESP32 device with an Android 9 or Android 11 device
2. Switch ESP32 device off
3. Switch ESP32 device on
4. Android device tries to reconnect, ESP32 device detects it, but the connection fails.

Using the lib version without NimBLE, ESP32 device reconnects perfectly.

CybershoesVR, 2022-10-17:
Today we have the same issue as @remsurge 

**It worked on July 7th 2022 with the following configuration:**
Meta Quest2 + ESP32-C with libraries NimBLE 1.38 and ESP32-BLE-gamepad 4.1

Meanwhile it does not work anymore with the same constellation. Obviously Meta has updated their FW.

**Futher observations:**
@lemmingDev We tested the provided zip file. Result: can complile & connect, but gamepad is disfunctional.
NimBLE 1.4  with ESP32-BLE-Gamepad 5.2 Result: ESP32 CRASHES
NimBLE 1.4 with ESP32-BLE-Gamepad 5.1 Result: ESP32 CRASHES

NimBLE 1.38 with ESP32-BLE-Gamepad 5.2 AutoReport: FALSE Result: connects & reconnects, but gamepad is disfunctional.
NimBLE 1.38 with ESP32-BLE-Gamepad 5.2 AutoReport: TRUE Result:  connects & reconnects, but gamepad is disfunctional.
NimBLE 1.38 with ESP32-BLE-Gamepad 5.0 AutoReport: FALSE or FALSE: only works the first time (and I've commented out the battery code as you did in version 5.1). It cannot be made working by forget device and pairing again.

### BEST RESULT:
NimBLE 1.38 with ESP32-BLE-Gamepad 5.1 AutoReport: TRUE or FALSE 
**Result: connects & reconnects after ESP32 powerloss. However: gamapad only works at first conection.**
**When hitting "Forget Device" and "Pair" again, Gamepad WORKS until next powerloss.**

_@lemmingDev you may DM dev@cybershoes.com if you need a Quest to debug. Please help :)_

**Parallel observation:**
Random chinese BT gamepad called GamepadPlusV3 works with Quest as it should

CybershoesVR, 2022-10-17:
PS: maybe related: https://github.com/h2zero/NimBLE-Arduino/issues/464

CybershoesVR, 2022-10-18:
I tried a few security settings but decided it'll be better to **debug** my Meta Quest2 with Oculus Developer Hub  https://developer.oculus.com/downloads/native-android/  , since I have a gamepad here that works as expected. 

from the log: 
**GamePadPlus V3** is a BT gamepad device that reconnects successfully.
**Cybershoes2** is our BT gamepad device that does reconnect but gamepad functionality fails upon reconnect.

What I did for each device was:
* pair & confirm
* power off the gamepad
* power on the gamepad
* power off the gamepad

_@lemmingDev / @remsurge, do you have an ideat what makes the difference? Respectively how to change the settings in BleGamepad.cpp > taskServer to make it work?_ 

Using NimBLE 1.38 with ESP32-BLE-Gamepad 0.5.1, because using NimBLE 1.40 still makes the ESP32-C crash

[odh_logs_2022-10-18 13.55.53.833.txt](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/9811240/odh_logs_2022-10-18.13.55.53.833.txt)

01GOD, 2023-07-11:
> I tried a few security settings but decided it'll be better to **debug** my Meta Quest2 with Oculus Developer Hub https://developer.oculus.com/downloads/native-android/ , since I have a gamepad here [...]

I've been trying to do BLE through Unity with an ESP32 and a Quest 1. I used the BLE for iOS and Android package. The example functions with Android phone, but only connects to Unity app on Quest. Basically the app says it can't connect to the BLE services. Saw this? https://communityforums.atmeta.com/t5/Quest-Development/Bluetooth-scan-and-permissions-BLE-and-the/m-p/970810/highlight/true#M4775

lemmingDev, 2024-02-24:
So, latest 5.4 version of this library seems to be working perfectly with Quest 2 and a regular ESP32, where it reconnects fine automatically

I've made a few small changes that may make it work with other ESP32 variants

I'll publish it soon. Please test it out and report back

lemmingDev, 2025-01-12:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.6.3 along with latest NimBLE might fix reconnection problems

Can anyone re-test and report back?

lemmingDev, 2025-01-24:
No recent feedback provided
Will close

closed 2025-01-24 by lemmingDev