lemmingDev/ESP32-BLE-Gamepad #137: Xinput support

hachirokumiku, 2022-10-09:
would you add xinput support to ESP32-BLE-Gamepad?

lemmingDev, 2022-10-09:
Feel free to get it working and do a pull request 😉

LeeNX, 2022-10-29:
Microsoft XInput, not to be confused with X.orgs XInput ... ;-). 
  
Found - https://github.com/dmadison/ArduinoXInput with something data info - https://www.partsnotincluded.com/understanding-the-xbox-360-wired-controllers-usb-data/. 
  
but I don't have MicroSoft Windows to test or play with, maybe somebody else might find some of this useful.  
  
But my thinking, is that you would need to license or emulate a Microsoft XBox controller.   
  
MS XInput device seems a little limited compared to the BLE HID Game device.

Mystfit, 2023-12-25:
I managed to get XInput working by emulating either a Xbox One S (model 1708) or Xbox Series X (model 1914) controller over in my composite HID fork. It does mean that the buttons and axis' end up having to be fixed rather than dynamic as I had to match up with what XInput expected via the HID report map.

I would like to backport the Xbox controller code but I've refactored a big chunk of the library to work with combining a mouse and keyboard into the same device so it might be tricky.

lemmingDev, 2024-02-24:
Legend
https://github.com/Mystfit/ESP32-BLE-CompositeHID

LeeNX, 2024-03-19:
Hey @Mystfit , your fork does not have open issues, where is the best place to chat about some suggestions. I don't really want to message from within this repo. Would make it a little noisy.  
  
I would like to suggest an update to the ReadMe in the default branch, so that people known what is different, without having to go a little deep diving.  
  
I was also wondering about how much overhead adding adding keyboard and mouse into the same library? Latency and Resources?

Mystfit, 2024-03-19:
> Hey @Mystfit , your fork does not have open issues, where is the best place to chat about some suggestions. I don't really want to message from within this repo. Would make it a little noisy.
>
> I [...]

Hi @LeeNX, thanks for the heads up - I've opened up issues and discussions on the fork and we can continue this conversation over there if needed.

I have an updated readme that I've been working on and forgot to check in that will clarify the differences between the libraries. The gamepad portion is relatively similar to this library but it is initialised differently - I'll make sure to make that the primary code example in the readme shows the difference. I have updated all of the included code examples in the library to work with the new system though.

The combined HID report descriptor has an upper limit of 512 bytes (NimBLE limitation) which can be exceeded if you combine too many devices together or enable too many features. I've managed to exceed it before so I need to create a descriptive error message and put this limitation in the readme.

LeeNX, 2024-03-19:
Thanks @Mystfit , that is awesome. I will move my questions and PR into your repo.  
  
@lemmingDev I think we can point Xinput and multiple device question into https://github.com/Mystfit/ESP32-BLE-CompositeHID repo. See if I can get around to a PR soon.

Thank you very much you awesome people!!

mmmrrjs, 2024-08-22:
> Thanks @Mystfit , that is awesome. I will move my questions and PR into your repo.
> 
> @lemmingDev I think we can point Xinput and multiple device question into https://github.com/Mystfit/ESP32-BLE-CompositeHID repo. See if I can get around to a PR soon.
> 
> Thank you very much you awesome people!!

It seems that this project can't running correctly？
I see #181 ，It can use some software（x360ce） to map to xbox controller
But i can't find anything like that on Android,so it seem hard to use gamepad on Android
If there was such a feature maybe the controller would be a lot easier to use
(I am sorry for my english it is not my native language.)

LeeNX, 2024-08-22:
@mmmrrjs if you looking for a BlueTooth XInput library, take a look at https://github.com/Mystfit/ESP32-BLE-CompositeHID. 
  
As for this project not running correctly, I am not sure what you mean or looking for?  
  
This Library provides a standard BLE Gamepad interface, that has many options, but are up to the host OS to interface and use them.  
  
XInput is a MicroSoft interface and is focused on providing XBox gamepad interface, which MicroSoft licenses and has not invested interested in making any other gamepad API public or usable.   
  
There are tools like https://ds4-windows.com, similar to Steam Input which can remap buttons and possible add macros, but I think you looking for a similar tool on Android, how about - https://play.google.com/store/apps/details?id=app.mantispro.gamepad&hl=en. Not an endorsement, as I have not used this application, just a possible suggestion.  
  
Another suggestion, best not to try and discuss close issues, unless you adding information, at least that is a rule of thumb I use.  
  
@mmmrrjs I hope you find a tool that fits your needs, best of luck.

LeeNX, 2025-01-09:
with [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID), is it worth keeping this open? I don't think that XInput should be implemented. @lemmingDev what do you think?

lemmingDev, 2025-01-12:
> with [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID), is it worth keeping this open? I don't think that XInput should be implemented. @lemmingDev what do you think?

Yeah - closed ;)

closed 2025-01-12 by lemmingDev