lemmingDev/ESP32-BLE-Gamepad #138: No Name in some Games

Foxloop, 2022-10-12:
I Have a problem with the name of the Controller. When i conneted to Windows everything seem ok. I Got a name of the Controller and See everything.

I Got the PID and the VID. This is OK. 

But some Games can See the Controller and i can play but Shows no Name from Controller, especially XInput Games.

Can you Tell how to configure this please?

Thanks a lot.

lemmingDev, 2022-10-22:
Read here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/77

Foxloop, 2022-10-22:
Thanks for your replay.

I make it in Windows but i Have still the same problem with the ble gamepad.

Sometimes when i start the game, i get hyroglifics in the Controller name.

I search in the Hid Descriptor for a solution but i cant find it.

I think the name comes from the Hid Descriptor, maybe the country flag or something like this.

But i am not such a professional programmer to find it.

Maybe you can explain it or you know how to change it. My thought is that the Descriptor give the Rawinput Name, when i get the hyroglifics.

In the older library i get always hyroglifics. I the New library i do t Have a rawinput joystick and no Name anymore, but in the Games if they read rawinput Name from BLE Gamepad i always have no Name or different Name. Not like in windows. Its a Little bit strange for me.

Maybe you can help.

closed 2024-02-24 by lemmingDev