lemmingDev/ESP32-BLE-Gamepad #139: Trying to send L1, L2, R1, and R2 buttons

1Yanik3, 2022-10-16:
Sorry in advance if this is a stupid question, but I have been trying for hours by now.

I have managed to successfully setup a controller where I can press buttons: A, B, X, Y, START, SELECT. As well as a D-PAD. That part works wonderfully and functions perfectly on my Android.

However, I have not managed to send L1, L2, R1, or R2. I've tried to send every BUTTON_6 through BUTTON_10, with none resulting on any kind of input on the device side (I am using a Gamepad Tester app, two actually with both not reporting anything).

So the question is: How can I send L1, L2, R1, and R2 presses?

Thank you for any help in advance.

LeeNX, 2022-10-16:
L2 and R2 are often used as Accelerator and/or Brake and not a digital button, but a trigger/slider.

Thou if you looking at https://w3c.github.io/gamepad/#remapping, L1 (buttons[4]) and R2 (buttons[5]), I am not sure how these maps to button in the BLE setup. Could also be how the OS defines and passes the gamepad setup to the browser.

I would test by running https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/TestAll/TestAll.ino and monitoring the serial feedback and see if ```ESP32-BLE-Gamepad``` at least triggers the buttons you want.

An idea, is L2 could possible be done ```bleGamepad.setAccelerator(i);```, as seen https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/DrivingControllerTest/DrivingControllerTest.ino#L99

lemmingDev, 2024-02-24:
On Windows, L1 and R1 are probably buttons 5 and 6
L2 and R2 are probably left trigger and right trigger

It's all OS dependent

closed 2024-02-24 by lemmingDev