lemmingDev/ESP32-BLE-Gamepad #141: [Feature Request] Add motion sensor support

LeeNX, 2022-10-22:
More a place holder for info to get motion support into the library.  
  
Extend the library to support motion feedback, much like PS5 Dual sense controller, basically reporting something like 3 axis accelerometer and/or 3 axis gyroscope.  
  
I have seen info on HID/GATT characteristics and will follow up with more info, but possible?
Sensors Page (0x20)
Usage ID:
 * 71 Motion: Accelerometer 1D
 * 72 Motion: Accelerometer 2D
 * 73 Motion: Accelerometer 3D
 * 74 Motion: Gyrometer 1D
 * 75 Motion: Gyrometer 2D
 * 76 Motion: Gyrometer 3D

lemmingDev, 2022-10-22:
I guess it's possible
Are there any HID examples to glean the information we need?

LeeNX, 2022-10-23:
Thanks @lemmingDev 

I don't think the feature is a high demand feature. Basically got an IMU and saw a video giving me an idea - https://youtu.be/3qlZmXnE1mw. 
  
Only two user cases I could think of, as per the video, aim assist/direction from motion movement. Another would be steering wheel like the Wii did it, without using the direction hat or joysticks.  
  
No HID examples, other than what PS4/5 or possible XBox controller, but they looked very none standard.  
  
The other thing that comes to mind, is what would use this feature, ie: SDL or picked up by OS

Utility like ```JoyShockMapper``` https://github.com/Electronicks/JoyShockMapper could be useful for insights, but could add latency or would just be added directly into the ESP32 firmware used.  
  
If I find more info, I will added. Welcome to add ```needs-more-info``` tag or something.

LeeNX, 2022-10-31:
Video of somebody using gyro aim - https://youtu.be/njIj-I3VPZg?t=814

lemmingDev, 2024-02-24:
We could map a gyroscope to any of the existing axes with code without modification to the library.
Simply map the values to existing axes

I guess we could add it as an individual thing like with the simulation controls

Going through all of the old issues

You still want me to look into it?

LeeNX, 2024-02-26:
oh I see @lemmingDev cleaning out all them old issues, thanks!

Mapping to a set of standard axes could be done, thou I don't know what that math would look like.  
  
I was thinking more about exposing like gyro aiming assist or something like - https://help.rewasd.com/how-to-remap/gyroscope.html

Then it would be up to the host device to do as it needs with the IMU readings.  
  
I don't know what Windows/OSX would do with these ```characteristic```, but Linux and ```bluez```, I think it would be possible to get updates. I was looking at what the DS/3/4/5 controllers were doing, but need to finish that research.

LeeNX, 2024-03-19:
Greetings @lemmingDev , hoping I could ask for assistance with adding a ```characteristic``` without changes to the library, like an extension or override as an example. I would then be able to poke at other ideas without big changes. I just can't get my code to work, but that is my poor C skills.

zeknoss, 2024-11-30:
Hey @LeeNX , did you have any progress in your venture?

LeeNX, 2024-12-01:
I have to admit, I am not sure where I put this sensor. My focus has been on other projects where ESP32-BLE-Gamepad will be useful. I currently don't have a use case for IMU sensors in antBot, thou if anybody can think of a way it could be used to be an advantage, I am all ears. Only thing worse than my coding skills is my battle bot controls skills. '-)

zeknoss, 2024-12-01:
> I have to admit, I am not sure where I put this sensor. My focus has been on other projects where ESP32-BLE-Gamepad will be useful. I currently don't have a use case for IMU sensors in antBot, thou [...]

I'm currently developing a wireless head tracker for use in simulation games.
I've managed to translate the LSM6DSM's gyro and accelerometer data to thumbstick outputs and I'll try them today in a game (Elite Dangerous or X4: Foundations) to test how the cockpit view works with joypad axis.

I'll let you know if you want :)

lemmingDev, 2025-01-12:
> I'm currently developing a wireless head tracker for use in simulation games. I've managed to translate the LSM6DSM's gyro and accelerometer data to thumbstick outputs and I'll try them today in a [...]

How did this go?

zeknoss, 2025-01-12:
> How did this go?

Well, I got a project and wasn't able to finish it yet :( Once I do, I will update here.

lemmingDev, 2025-02-01:
Ooh - I just added motion support using desktop page versions without referring to this post with sensors page

I will quickly try these ones out and see how they go

LeeNX, 2025-02-01:
> Tries the 0x20 sensor usage page with 3 axis gyro and 3 axis accel and got driver error

On Windows or other host devices or OS? Looking at a few other HID, not seen any that use a standard usage page, normally all  devices use a custom usage page.

For example - https://gist.github.com/ToadKing/b883a8ccfa26adcc6ba9905e75aeb4f2
which decoded - https://gist.github.com/rossmacarthur/64475d9ce3350d99b49203731dc7aa0e

> For now it's other axes

Would you mind pushing a branch with your work-in-porgress with the 0x20 sensor usage page. Could be something that I could look over and possible use for a base.

I did not know as much about BLE devices and how host OS are needed to support the different kind of devices when I opened this feature request. Would be nice, but I don't think many, if any, of the host OS will have support for this. I don't have the skill to get support into other tools or host like Linux, still working on the PlayerLED stuff, which I think has more use or part of a stepping stone type thing.

lemmingDev, 2025-02-01:
Yeah - I don't think 0x20 sensor usage page should really be used for a gamepad

I think if we wanted to have it plug and play, and not eat up 6 out of the 8 available axes, we'd need to emulate a ps3 or 4 dual shock or something with custom pages that uses a driver or companion app

I think its not really in our scope, though maybe in a fork, similar to the xinput one

I deleted the page 0x20 test code already, but its super simple to modify the one file again

I tested two different types:

```
  // Accelerometer3D
  0x05, 0x20,       // Usage Page (Generic Desktop)
  0x09, 0x73,       // Usage (Accelerometer3D)
  0xA1, 0x00,       // Collection (Physical)
  0x75, 0x10,       // Report Size (16 bits)
  0x95, 0x03,       // Report Count (3 for X, Y, Z)
  0x16, 0x00, 0x80, // Logical Minimum (-32768)
  0x26, 0xFF, 0x7F, // Logical Maximum (32767)
  0x81, 0x02,       // Input (Data, Variable, Absolute)
  0xC0,             // End Collection

  // Gyrometer3D
  0x05, 0x20,       // Usage Page (Generic Desktop)
  0x09, 0x76,       // Usage (Gyrometer3D)
  0xA1, 0x00,       // Collection (Physical)
  0x75, 0x10,       // Report Size (16 bits)
  0x95, 0x03,       // Report Count (3 for X, Y, Z)
  0x16, 0x00, 0x80, // Logical Minimum (-32768)
  0x26, 0xFF, 0x7F, // Logical Maximum (32767)
  0x81, 0x02,       // Input (Data, Variable, Absolute)
  0xC0,             // End Collection
```
and also
```
   // Accelerometer Collection (3D Accelerometer)
    0xA1, 0x00,        // Collection (Physical)
    0x05, 0x20,        // Usage Page (Sensors)
    0x09, 0x71,        // Usage (Motion: Accelerometer 1D)
    0x09, 0x72,        // Usage (Motion: Accelerometer 2D)
    0x09, 0x73,        // Usage (Motion: Accelerometer 3D)
    0x16, 0x01, 0x80,  // Logical Minimum (-32767)
    0x26, 0xFF, 0x7F,  // Logical Maximum (32767)
    0x75, 0x10,        // Report Size (16 bits)
    0x95, 0x03,        // Report Count (3 axes)
    0x81, 0x02,        // Input (Data, Var, Abs)
    0xC0,              // End Collection (Accelerometer)

    // Gyroscope Collection (3D Gyroscope)
    0xA1, 0x00,        // Collection (Physical)
    0x05, 0x20,        // Usage Page (Sensors)
    0x09, 0x74,        // Usage (Motion: Gyrometer 1D)
    0x09, 0x75,        // Usage (Motion: Gyrometer 2D)
    0x09, 0x76,        // Usage (Motion: Gyrometer 3D)
    0x16, 0x01, 0x80,  // Logical Minimum (-32767)
    0x26, 0xFF, 0x7F,  // Logical Maximum (32767)
    0x75, 0x10,        // Report Size (16 bits)
    0x95, 0x03,        // Report Count (3 axes)
    0x81, 0x02,        // Input (Data, Var, Abs)
    0xC0,              // End Collection (Gyroscope)
```

lemmingDev, 2025-02-01:
ok - here is the updated BleGamepad.cpp

[BleGamepad.zip](https://github.com/user-attachments/files/18630495/BleGamepad.zip)

Test with latest master files and https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/MotionController/MotionController.ino