lemmingDev/ESP32-BLE-Gamepad #142: [Feature Request] Branch per BLE stack?

LeeNX, 2022-10-23:
Maybe more a suggestion that a feature request, possible to keep both ```NimBLE``` and ```Bluedroid``` branches on github? I see that NimBLE is in place, but using zip archives files for the Bluedroid stack in issues, must add unneeded overhead for the developer. Switching between branches locally should be easy, thou I am no git guru.

lemmingDev, 2022-10-23:
Yeah - was going to do that, but ran out of time/motivation last holidays after making an updated Bluedroid version for a few people who were chasing it

LeeNX, 2023-07-06:
I was poking at the [T-vK/ESP32-BLE-Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard) and saw that it supports both libs with ```#if``` defines, like - https://github.com/T-vK/ESP32-BLE-Keyboard/blob/f8dd4852113a722a6b8dc8af987e94cf84d73ad5/BleKeyboard.cpp#L3. 
  
I was thinking possibly three branchs, like main that supports both with the define and two cleaner branchs for each lib. It's a bit more work, but gives us all three options.  
  
See what I can hack up, soon.

lemmingDev, 2023-07-06:
Interesting

Any help appreciated!

LeeNX, 2025-02-01:
I think we can close this issue too, @lemmingDev what do you think?

lemmingDev, 2025-02-01:
Yeah

closed 2025-02-01 by lemmingDev

LeeNX, 2025-02-20:
Just for reference - https://github.com/espressif/arduino-esp32/discussions/10991#discussioncomment-12259976

lemmingDev, 2025-02-20:
Thanks for the heads up