lemmingDev/ESP32-BLE-Gamepad #144: Some questions about setting the battery level

qyxa007, 2022-10-29:
Hello! I used this library today. It's very good! Also, I notice that in "BleGamepad.cpp", the code in the "setBatteryLevel" function is all commented out. I hope you can add this function, thank you!You can contact me by E-mail :ttxl3099@qq.com, thank you!Looking forward to your E-mail.

lemmingDev, 2022-10-29:
Hi

It's commented out because it's not supported by the currently released version of NimBLE

Once they release a new version, I'll test it and enable it

LeeNX, 2022-10-29:
Update the NimBLE library with https://github.com/h2zero/NimBLE-Arduino/commit/1fcc3c396c17beabae3458a8cd10a6026dee3a45 patch and then uncomment the section in the ESP32-BLE-Gamepad library.

I can confirm this works fine.

LeeNX, 2022-10-29:
haha @lemmingDev beat me to the comment.

LeeNX, 2022-10-29:
NimBLE version [1.4.1] - 2022-10-23 is out, but their releasing process is a little harder to track - https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/CHANGELOG.md#141---2022-10-23

lemmingDev, 2022-10-29:
Seems to compile fine with the setBatteryLevel function enabled in conjunction with NimBLE 1.4.1
Will test it on an ESP32 tomorrow, and if it works, I'll release an update

qyxa007, 2022-10-29:
After I tried nimble1.4.1, it seemed to work well. Here are some pictures and videos - I'm from China, so my computer is in Chinese, but you can use the translation software to read and understand the Chinese.Perhaps this passage is a little strange to read in English? That's because I used translation software, because my English is not very good...haha.

https://user-images.githubusercontent.com/95993288/198839977-b46467dd-4b12-4c16-ae90-413877e4c3f1.mp4

[image]
[image]

qyxa007, 2022-10-29:
After I tried **_nimble1.4.1_**, it seemed to work well. Here are some pictures and videos - I'm from China, so my computer is in Chinese, but you can use the translation software to read and understand the Chinese.Perhaps this passage is a little strange to read in English? That's because I used translation software, because my English is not very good...haha.

https://user-images.githubusercontent.com/95993288/198840428-c082128f-e65a-42b6-90cf-82b1eb2ed862.mp4
[image]
[image]

TTXL3099, 2022-10-29:
> Update the NimBLE library with [h2zero/NimBLE-Arduino@1fcc3c3](https://github.com/h2zero/NimBLE-Arduino/commit/1fcc3c396c17beabae3458a8cd10a6026dee3a45) patch and then uncomment the section in the ESP32-BLE-Gamepad library.
> 
> I can confirm this works fine.

Thanks for your reply. This really solves the problem!

TTXL3099, 2022-10-29:
I also want to know the difference between **_CONTROLLER_TYPE_JOYSTICK, CONTROLLER_TYPE_GAMEPAD and CONTROLLER_TYPE_MULTI_AXIS_**. Can you tell me?

LeeNX, 2022-10-30:
From the ```HID Usages and Descriptions``` PDF file - https://usb.org/sites/default/files/hut1_3_0.pdf. 
  
Application Usages:
 - Joystick
 ```
 A manual control or cursor device. A joystick minimally consists of two variable
axes (X and Y) and two buttons. A joystick is typically a rotational motion sensor.
However, for legacy reasons, it is defined using linear axes.
Traditionally, a joystick driver applies its own scaling to values returned
from a joystick. That is, the driver simply linearizes and translates the range of
values generated by the stick into normalized values between 0 and 64K, where
32K is centered. The application (game) then interprets the normalized values as
necessary. Because of this, joysticks normally do not declare Units or Physical
Minimum and Physical Maximum values for their axes. Depending on the
driver, these items may be ignored if they are declared.
 ```
 - Gamepad
 ```
 A manual control or cursor device. A game pad minimally consists of a
thumb-activated rocker switch that controls two axes (X and Y) and has four
buttons. The rocker switch consists of four contact closures for up, down, right,
and left.
 ```
 - Multi-axis Controller
 ```
 An input device used to orient eyepoints and or objects in 3 dimensional space. A
Multi-axis Controller typically consists of six, variable axes (X, Y, Z, Rx, Ry and Rz)
and is used by CAD/digital content creation applications for model manipulation
and visualization in 3D space. The device may incorporate zero or more buttons.
 ```
  
Basically up to the OS to interpret what HID Application Usage is declared or how the OS might present the BLE device as an input.

I don't believe that the ESP32-BLE-Gamepad library does anything with this values, another than put them into the HID descriptor.

lemmingDev, 2022-11-01:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.5.3

Latest version re-enables the setting of battery level

closed 2022-11-01 by lemmingDev

lemmingDev, 2022-11-10:
Hi

This was a simplistic example to show the syntax

It's up to you to decide where best to call it