lemmingDev/ESP32-BLE-Gamepad #145: Is it compatible with vibrations motors ?

Pyloupy, 2022-10-30:
Hey, i discover this Amazing project and esp32 a  little while ago and  i have a question:
Can this library support vibrations motors like in traditionnal controller (Xbox or PS4) and if yes how to add this function ?
Thanks
(Sorry for my really bad english im french and 14 years old)

LeeNX, 2022-10-30:
Duplicate of https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/99  
  
We are discussing and a little progress been made in https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/43 
 
 Any assistance is welcome.

lemmingDev, 2024-02-24:
There is a fork of the library that does it here
https://github.com/Mystfit/ESP32-BLE-CompositeHID