lemmingDev/ESP32-BLE-Gamepad #148: Latency Measurements

AaronWebster, 2022-11-15:
Has anyone attempted to measure the input latency of this implementation?

LeeNX, 2022-11-16:
Something I have been thinking about doing, mostly because I don't know how good my coding or main loop would be.  
  
I see the [GP2040](https://github.com/FeralAI/GP2040) / [GP2040-CE](https://github.com/OpenStickFoundation/GP2040-CE) had some timing information in their project(s) -  https://inputlag.science/controller/methodology 
  
But I have not yet setup any tooling for tests. If you have any suggests or PR, I think they would be welcome.  
  
Would be really cool if we have something like a GitHub Action that test per release or something.

lemmingDev, 2023-05-18:
Thanks for pitching in @LeeNX
 
Let me know if anything becomes apparent from the testing. 

Perhaps some tweaks to the code or some better examples are needed

jlarnal, 2024-03-12:
Ah... nothing since last year ?

LeeNX, 2026-09-18:
In another related project, I have done some basic latency measurements, but only of the BLE to host, not a gamepad implementation.

This can be found at [ESP32-BLE-Gamepad-HIL v0.2.5](https://github.com/LeeNX/ESP32-BLE-Gamepad-HIL/releases/tag/v0.2.5) - blatant plug of my own project.

There are a number of other things that I can been thinking of that might be something to think about, as not all implementations are equal. Think of button debounce vs over-sampling of analog readings.

I don't think there is much more to really discussed, but I think this can be closed, unless somebody wants to jump in and add any other information.

lemmingDev, 2026-09-18:
I guess it would be useful to publish, if known, the time in ms it takes from the Arduino code sending the button press call until the time the OS actions it for generic, SInput and XInput

Maybe an advertised polling rate, though given its BLE and we generally only send reports on state changes, it may not be as relevant, but might be nice to know

LeeNX, 2026-09-19:
My initial through on the matter, was that a smaller HID description, ie less buttons and inputs, would mean faster or more inputs per second than a larger HID description, but my HIL profile minimal vs maxbtn did not seem to yield any noticeable difference, at least from triggering inputs to seeing the inputs on the HIL host. 

M thinking was the larger data set would need more bluetooth radio time and therefor less input, but it seems like the same number of inputs. This could also be because the difference between small HID of 64bytes vs large HID of 134bytes is not enough to show a difference.

But this could also mean, the difference between collecting all input and push one change over the BLE radio vs firing many smaller changes have an impact.

The more I think about it, the more I think I don't really understand all the possible optimisations that could be done or even how it all works.

As for the different reports between generic vs SInput vs XBox modes, could show differences in implementations of drivers or even layers of access between application and end devices (gamepad). I think we will only see this Linux, unless somebody can run the tests for OSX and/or Windows. Again, drivers could be the interesting corner case.

So many questions and as I think I was pointing out, am I even asking the right questions. This is where I am hoping people who know more or better can point out where there might be misunderstandings.

lemmingDev, 2026-09-21:
Interesting findings how more data in the report doesn't seem to slow it down.

Would be nice to publish a set of numbers for the default sketch for each controller type and call it a day....

LeeNX, 2026-09-22:
Still not sure if my findings are correct, would be nice if somebody else double checked and confirmed or even pointed out anything that is not right.

Working on a slim/minimal HID report size vs a silly huge HID report size. Also other OSs and BlueTooth adapters all could be different. Lots of moving parts I think. I am not sure that anybody is going to be making an eSport gaming controller or at least not for something that needs or wants kHz polling, thou I think that is just marking speak. 

Thou looking over https://gamepadla.com, there are some bad wireless gamepad dressed up to be awesome.

I am working on a BluePad32 HIL too, so that I can do more testing too. Also working on an SInput DKMS driver https://github.com/LeeNX/linux-hid-sinput that I want to put through it's paces and see if it works better than HIDApi for SDL. Also interesting numbers to look over.

lemmingDev, 2026-09-22:
Nice

On the topic of SInput, you'll be able to test with GP2040-CE SInput implementation too if you don't have an rp2024 board to play with, as I'm working on porting it to ESP32-S3 (and maybe ESP32) and I know you've got one

I've got it up and running with almost feature parity at the moment for working as wired gamepads, and have the webconfig working over WiFi
I'm yet to add any Bluetooth gamepad output yet (or input, as that might be interesting), as I'm still polishing off wired parity. Once parity is finished, I'm going to look at adding Bluetooth stuff

I'm not sure if anyone will be interested

I'll hide the announcement in here since nobody will read it - lol

LeeNX, 2026-09-22:
Interesting, I have some rp20xx boards some place, but better know the ESP32 dev boards, so happy to play there at the moment. The SInput kernel drive might be useful, but if I remember correctly, you running Windows, so not sure if that is of any help.

Best of luck with the rp20xx stuff, when I suggested on BLE work on the GP2040-CE repo, it got shot down pretty quickly, but they might have had focus on something else.

I am very keen on wireless tools, always worry somebody going to pull a TV off the wall playing games, wireless gamepads have less of that problem, but some other problems. ;-)

I hope other people find something interesting to play with between all the foundation and testing been setup.