lemmingDev/ESP32-BLE-Gamepad #149: Not working with Recalbox (And Retropie ?) on Raspberry pi(4)

Leogenie, 2022-11-18:
Hello,

Excuse my bad English.
My Gamepad (2 axis x +y and 12 buttons) detected in recalbox and associated, but in key configuration menu, no keys or axis are detected.

Tanks

LeeNX, 2022-11-19:
Can you get access to a shell on your R-Pi? Either via ssh into the device or by quitting ```Emulation Station``` without powering off the R-Pi.  
  
Then you can check if the gamepad is registered with the input system with 
```bash
ls -la /dev/input/
```
Should see something like 
```
leet@thor:~ $ ls -la /dev/input/
total 0
drwxr-xr-x   3 root root     160 Nov 19 09:37 .
drwxr-xr-x  17 root root    4200 Nov 19 09:37 ..
drwxr-xr-x   2 root root      80 Nov  8 15:25 by-path
crw-rw----   1 root input 13, 64 Nov  8 15:25 event0
crw-rw----   1 root input 13, 65 Nov  8 15:25 event1
crw-rw----+  1 root input 13, 66 Nov 19 09:37 event2
crw-rw----+  1 root input 13,  0 Nov 19 09:37 js0
crw-rw----   1 root input 13, 63 Nov  8 15:25 mice
```
then using something like jstest, you can see the details and some testing
```bash
leet@thor:~ $ jstest --normal /dev/input/js0
Driver version is 2.1.0.
Joystick (ESP32TestOrg) has 9 axes (X, Y, Z, Rx, Ry, Rz, Throttle, Hat0X, Hat0Y)
and 64 buttons (BtnA, BtnB, BtnC, BtnX, BtnY, BtnZ, BtnTL, BtnTR, BtnTL2, BtnTR2, BtnSelect, BtnStart, BtnMode, BtnThumbL, BtnThumbR, ?, (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null), (null)).
Testing ... (interrupt to exit)
Axes:  0:     0  1:     0  2:     0  3:     0  4:     0  5:     0  6:     0  7:     0  8:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off 13:off 14:off 15:off 16:off 17:off 18:off 19:off 20:off 21:off 22:off 23:off 24:off 25:off 26:off 27:off 28:off 29:off 30:off 31:off 32:off 33:off 34:off 35:off 36:off 37:off 38:off 39:off 40:off 41:off 42:off 43:off 44:off 45:off 46:off 47:off 48:off 49:off 50:off 51:off 52:off 53:off
Axes:  0:     0  1:     0  2:     0  3:     0  4:     0  5:     0  6:     0  7:     0  8:     0 Buttons:  0:off  1:on   2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off 13:off 14:off 15:off 16:off 17:off 18:off 19:off 20:off 21:off 22:off 23:off 24:off 25:off 26:off 27:off 28:off 29:off 30:off 31:off 32:off 33:off 34:off 35:off 36:off 37:off 38:off 39:off 40:off 41:off 42:off 43:off 44:off 45:off 46:off 47:off 48:off 49:off 50:off 51:off 52:off 53:off
```

I have had problems, where my mobile device keeps stealing the gamepad connection from my R-Pi, so I have to turn off bluetooth on my mobile device. Disconnect is not enough. Or remove the paring from my mobile device.

I also had pairing issues on the R-Pi, when my mobile or desktop kept resetting or pairing to my gamepad while I was trying to pair  with my testing R-Pi. Was very frustrating.   
  
Some in progress notes, that might help: https://github.com/LeeNX/ESP32-BLE-Gamepad/tree/test-led#troubleshooting

Leogenie, 2022-11-19:
Hi,

Tanks for your help.

"ls -la /dev/input/" give me 2 joy js0 and js1 (+ mice and other things)

"jstest --normal /dev/input/js0" don't work and sdl2-jtest or sdl-jtest not more. (Recalbox 8.??)

I think they'r Logitech USB and 8bitdo Bluetooth gamepads. (I have reset bluetooth gamepads, but js1 always present.
When i connect ESP32-Gamepad, i havn't 3 Gamepads...

with the command "bluetoothctl pair EC:94:CB:69:EF:xx && bluetoothctl connect EC:94:CB:69:EF:xx" (This is my mac esp32 adress) give
Attempting to pair with EC:94:CB:69:EF:52
[CHG] Device EC:94:CB:69:EF:52 Paired: yes
Pairing successful
Attempting to connect to EC:94:CB:69:EF:52
Connection successful

But don't OK.

And with usb logitech and ESP Gamepad connected :
the command "sdl-jstest --list" give
Found 1 joystick(s)

Joystick Name:     'Logitech Logitech Cordless RumblePad 2'
Joystick Number:    0
Number of Axes:     4
Number of Buttons: 12
Number of Hats:     1
Number of Balls:    0

And with USB Logitech, 8bitdo SN30 pro Bluetooth ans ESP Gamepad :
 sdl-jstest --list
Found 2 joystick(s)

Joystick Name:     'Logitech Logitech Cordless RumblePad 2'
Joystick Number:    0
Number of Axes:     4
Number of Buttons: 12
Number of Hats:     1
Number of Balls:    0

Joystick Name:     '8Bitdo SN30 Pro'
Joystick Number:    1
Number of Axes:     6
Number of Buttons: 16
Number of Hats:     1
Number of Balls:    0

The ESP32 Gamepad never appear.

The command "ls -la /dev/input/" list 3 gamepads js0, js1 and js2....
but "sdl-jstest --list" just 2 gamepads.

Sorry but my linux experiences stop at this point !
I can'n more understand at this time.

Yann

LeeNX, 2022-11-20:
A few times I have run into problems, where an older connected device (like my desktop), not even connected, would stop my R-Pi and Gamepad from re-connecting. I had to turn off all my Bluetooth hosts other than my R-Pi, else I would see the following in ```bluetoothctl```
```
leet@thor:~ $ sudo bluetoothctl
Agent registered
[CHG] Device 34:94:54:35:1D:F6 Connected: yes
[CHG] Device 34:94:54:35:1D:F6 Connected: no
[bluetooth]# devices
Device 34:94:54:35:1D:F6 ESP32TestOrg
[CHG] Device 34:94:54:35:1D:F6 Connected: no
[CHG] Device 34:94:54:35:1D:F6 Connected: yes
[CHG] Device 34:94:54:35:1D:F6 Connected: no
[CHG] Device 34:94:54:35:1D:F6 Connected: yes
[CHG] Device 34:94:54:35:1D:F6 Connected: no
[CHG] Device 34:94:54:35:1D:F6 Connected: yes
```
I recommend turning off other bluetooth hosts and/or remove pairing to other devices for your gamepad.  
  
Some other debugging that I found helpful, was running ```sudo btmon```, which gives you lots of bluetooth debugging on your R-Pi.  
  
Also checking the host to gamepad status in bluetoothctl
```
[ESP32TestOrg]# info 34:94:54:35:1D:F6
Device 34:94:54:35:1D:F6 (public)
	Name: ESP32TestOrg
	Alias: ESP32TestOrg
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:vE502pABCDd1001
	Battery Percentage: 0x19 (25)
```

I like to use ```sudo evtest``` to list input devices.
```
leet@thor:~ $ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	vc4
/dev/input/event1:	vc4
/dev/input/event2:	ESP32TestOrg
``` 
and then to check setup
```
leet@thor:~ $ evdev-joystick --s /dev/input/event2
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x01 (1) (Y Axis) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x02 (2) (Z Axis) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x03 (3) (X Rate Axis) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x06 (6) (Throttle) (value: 0, min: 0, max: 32767, flatness: 2047 (=6.25%), fuzz: 127)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
```
There is a possibility that you don't have permissions to access the gamepad path
```
leet@thor:~ $ ls -la /dev/input/
total 0
drwxr-xr-x   3 root root     160 Nov 20 10:59 .
drwxr-xr-x  17 root root    4200 Nov 20 10:59 ..
drwxr-xr-x   2 root root      80 Nov  8 15:25 by-path
crw-rw----   1 root input 13, 64 Nov  8 15:25 event0
crw-rw----   1 root input 13, 65 Nov  8 15:25 event1
crw-rw----+  1 root input 13, 66 Nov 20 10:59 event2
crw-rw----+  1 root input 13,  0 Nov 20 10:59 js0
crw-rw----   1 root input 13, 63 Nov  8 15:25 mice
```
In my case, I added my user to the ```input``` group, else only the user or accounts in the ```input``` group can access the devices. That might be why you can see the ```/dev/input/js2``` path, but can't not access it? Bit of a guess. You might notice that some of my commands have ```sudo``` in front to get around some security practices. An idea for testing, but not good for long term usage.  
  
I remember seeing some udev additions suggested by ```8bitdo```, which would help, but target only at ```8bitdo``` devices. This can also be done with udev rules or start-up scripts. 

I hope this is helpful and you find a solution.

Leogenie, 2022-11-20:
Hello,

I disconnect the gamepad and diseable my PC Bluetooth.
I upload in ESP32 the example "Gamepad" in ESP32 for test, and change loop script for press button 1 2s and release 1s.
After connection with Recalbox, The IDE monitor give me "Press buttons 5, 16 and start. Move all enabled axes to max. Set DPAD (hat 1) to down right." (I don't change the text !! But its Button 1 press and release.)
So the ESP gamepad considered it's connected.

[with: "# bluetoothctl"
Agent registered
[CHG] Controller B8:27:EB:CF:78:2A Pairable: yes
[ESP32 BLE Gamepad]#]

the first "info" give Paired: no. I type "pair" and after, "info" return:

[ESP32 BLE Gamepad]# info
Device EC:94:CB:69:EF:52 (public)
        Name: ESP32 BLE Gamepad
        Alias: ESP32 BLE Gamepad
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:vE502pBBABd1001
        Battery Percentage: 0x64 (100)

 But in Recalbox "Gamepad configuration" first choice ine Gamepad menu ("Configurer une manette" in french) no Keypress appear...

With "# evtest"
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      HJZ Mayflash Wiimote PC Adapter
/dev/input/event1:      Logitech Logitech Cordless RumblePad 2
/dev/input/event2:      HJZ Mayflash Wiimote PC Adapter Mouse
/dev/input/event3:      HJZ Mayflash Wiimote PC Adapter Consumer Control
/dev/input/event4:      vc4
/dev/input/event5:      vc4
Select the device event number [0-5]:

No BLE Gamepad.

"evdev-joystick --s /dev/input/event2" don't work, command not found.

"ls -la /dev/input/"
# ls -la /dev/input/
total 0
drwxr-xr-x    4 root     root           280 Nov 20 11:26 .
drwxr-xr-x   14 root     root          3760 Nov 20 11:26 ..
drwxr-xr-x    2 root     root           160 Jan  1  1980 by-id
drwxr-xr-x    2 root     root           200 Jan  1  1980 by-path
crw-rw----    1 root     input      13,  64 Jan  1  1980 event0
crw-rw----    1 root     input      13,  65 Jan  1  1980 event1
crw-rw----    1 root     input      13,  66 Jan  1  1980 event2
crw-rw----    1 root     input      13,  67 Jan  1  1980 event3
crw-rw----    1 root     input      13,  68 Jan  1  1980 event4
crw-rw----    1 root     input      13,  69 Jan  1  1980 event5
crw-rw-r--    1 root     input      13,   0 Jan  1  1980 js0
crw-rw-r--    1 root     input      13,   1 Jan  1  1980 js1
crw-rw----    1 root     input      13,  63 Jan  1  1980 mice
crw-rw----    1 root     input      13,  32 Jan  1  1980 mouse0

Tanks for your Help, but i don't find the solution yet...
And for info, sudo don't work on my Recalbox, Command not found !! Because i was already connected as root ?

LeeNX, 2022-11-20:
Bluetooth seems to be pairing and is connect, at least by the info provided.  
  
Looks like the device is not getting registered with linux input system, else I would expect evtest to list ```ESP32 BLE Gamepad```, but it's not there.  
  
I have not used ```Recalbox```, I am using Raspberry Pi Debian Bullseye. It's possible that recalbox has some customisation for other BlueTooth controllers, like XBox that is causing a problem. If you have access to syslogs or something, we can try further troubleshooting.

Running ```dmesg --follow``` and connecting my gamepad, I see something like the following
```
[1035230.895319] input: ESP32TestOrg as /devices/virtual/misc/uhid/0005:E502:ABCD.0198/input/input409
[1035230.900320] hid-generic 0005:E502:ABCD.0198: input,hidraw0: BLUETOOTH HID v10.01 Gamepad [ESP32TestOrg] on dc:a6:32:c2:ae:71
[1035441.089437] input: ESP32TestOrg as /devices/virtual/misc/uhid/0005:E502:ABCD.0199/input/input410
[1035441.094420] hid-generic 0005:E502:ABCD.0199: input,hidraw0: BLUETOOTH HID v10.01 Gamepad [ESP32TestOrg] on dc:a6:32:c2:ae:71
```
Did you do any customisation for your 8Bitdo or Logitech devices? Are your other controller devices BlueTooth devices too?  
  
Sorry, I don't have any other ideas. If I think of anything else, I will post.

Leogenie, 2023-01-30:
Hello,

J'avais abandonné !! Désolé. (I had given up!! Sorry.)

"dmesg --follow" don't work. "dmesg -c" to clear buffer and "dmesg" ....
2 first lines : ESP-32 Gamepad connection.
3 last lines : My 8bitdo gamapad connection.

[16288.790183] Bluetooth: hci0: security requested but not available
[16288.940942] Bluetooth: hci0: security requested but not available
[16364.184266] hid-generic 0005:2DC8:6101.0008: unknown main item tag 0x0
[16364.184589] input: 8Bitdo SN30 Pro as /devices/platform/soc/fe201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:2DC8:6101.0008/input/input12
[16364.184843] hid-generic 0005:2DC8:6101.0008: input,hidraw6: BLUETOOTH HID v1.00 Gamepad [8Bitdo SN30 Pro] on b8:27:eb:cf:78:2a

If that can help you ?

lemmingDev, 2024-02-24:
No activity in over a year - will close

closed 2024-02-24 by lemmingDev