lemmingDev/ESP32-BLE-Gamepad #152: There's an error, it doesn't work, I can't fix it

2575156125, 2022-12-15:
`#include <NimBLEDevice.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
*** [.pio\build\esp32dev\lib444\ESP32-BLE-Gamepad\BleConnectionStatus.cpp.o] Error 1
*** [.pio\build\esp32dev\lib444\ESP32-BLE-Gamepad\BleGamepad.cpp.o] Error 1
.pio/libdeps/esp32dev/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp: In constructor 'BleGamepadConfiguration::BleGamepadConfiguration()':
.pio/libdeps/esp32dev/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:21:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                                                      _hardwareRevision("1.0.0")
                                                                               ^
.pio/libdeps/esp32dev/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:21:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
.pio/libdeps/esp32dev/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:21:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
.pio/libdeps/esp32dev/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:21:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
.pio/libdeps/esp32dev/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:21:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]`

lemmingDev, 2022-12-16:
Perhaps if you posted the code you're trying to compile...

2575156125, 2022-12-16:
`#include <Arduino.h>

void setup() {
delay(100);
}

void loop() {
 delay(100);
}`

2575156125, 2022-12-16:
I just added your library and tried to run it, above is my main file, I only have this file

closed 2022-12-16 by 2575156125

2575156125, 2022-12-16:
Is it for the arduino? I am running in vs code (I suspect my environment is wrong)