lemmingDev/ESP32-BLE-Gamepad #153: Not able to access right thumb stick and left DPAD.

Rajan732, 2022-12-22:
I want to make a controller like ps4. I was successful in giving the left thumb stick and other button inputs but I cannot access the right thumb stick and left DPAD. I have used all the functions and tried various configurations of code but still, I'm facing the same issue. When I give commands to use the right thumb stick (rX and rY) then also the left thumb stick moves. 

I would really appreciate it if anyone can help me with this.

LeeNX, 2022-12-22:
Greetings...

Three things come to mind, misconfiguration in (could you share a link to
your code?)

Platform differences and testing tools? Which platform (OS) and what
testing tools are you using?

Have you tried the test all buttons sketch -
https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/master/examples/TestAll

Or complete project -
https://github.com/palsayantan/BLE-Game-Controller/blob/main/ESP32_BLE_Controller/ESP32_BLE_Controller.ino

My TestAll buttons sketch runs fine on a Raspberry Pi and OSX using an sdl
test gamepad.

Might be worth suggesting process for testing, including some tools. Let me
see what I can do tomorrow.

Thanks
LeeT

Rajan732, 2022-12-23:
>

Rajan732, 2022-12-23:
I'm using the example codes of IndividualAxis.ino and TestAll.ino for testing with my ESP32 DEVKIT V1.
I even created my own sketch and tested the functions separately.
For now, I was planning to first see if all the gamepad functions are working. After this, I will add all the necessary buttons and joysticks.

Rajan732, 2022-12-23:
I saw some closed forums and noticed that there was an issue of not being able to access the triggers and in the comments, it was suggested to use BLE-Gamepad 0.5 and the latest version of NimBLE (1.4.1). I tried this also but in this case although my ESP32 is able to connect to my android phone as a Gamepad but is not giving any input signals.

LeeNX, 2022-12-23:
After totally breaking my testing environment, I re-downloaded BLE-Gamepad v0.5.3 and NimBLE v1.4.1. I used TestAll and used the above Android App for testing and I think it looks like all buttons doing stuff.
[image]

LeeNX, 2022-12-23:
Sorry, scrolls a little fast for TestAll, need to slow the button testing down and see if I can better select each test set. I will take a poke at this further, but I don't think there is a problem with the library.

Rajan732, 2022-12-23:
In the image you sent, the leftTrigger and rightTrigger are working. But in my case, only left axis leftX and leftY are showing.

[image]

Rajan732, 2022-12-23:
I changed my board to an ESP32-S and the code is working fine now. Thank you for your help and support @LeeNX

closed 2022-12-23 by Rajan732

LeeNX, 2022-12-23:
You're welcome, hope we can figure out why it's not working and others don't run into the same problem.  
  
Interesting, what did you have it set at? I am using a NodeCMU-S2.

Rajan732, 2022-12-23:
I didn't quite understand your question but I'm using an ESP32-S2-Saola-1

LeeNX, 2022-12-23:
You saying you changed your physical Esp32 board, not just the ArduinoIDE
definition?

What board were you using?

Rajan732, 2022-12-23:
I didn't change any Arduino IDE settings. I kept the board at "ESP32 Dev Module" for both the boards.

Previously I used:
https://www.electronicscomp.com/esp32-development-board-with-wifi-bluetooth-india?gclid=EAIaIQobChMIm5GpsM2P_AIVhwsrCh3_Hg-mEAQYASABEgIKpfD_BwE

Then I changed to:
https://robu.in/product/ai-thinker-nodemcu-32s-esp32-development-board-ipex-version/?pid=1304499