lemmingDev/ESP32-BLE-Gamepad #158: button map for different OS.

Kashyapkoshti, 2023-01-07:


Kashyapkoshti, 2023-01-07:
is there any easy way for selecting button mapping based on the which OS the device is connected?

LeeNX, 2023-01-07:
I don't think I have seen a BLE exposed idea of the connected server host OS. I could be wrong, if so, please let us know.  
  
I think I have seen a snippet (which I can't find at the moment) that could report ```server``` name and BLE MAC address.  
  
There are a few ways that I think you might be able to set button mappings from the ```BLE-Gamepad``` device/client, like check a button on start up/connect and switch to required mappings. Another way if you have spare GPIO, use a switch to select the mappings. Could also check the BLE server name/mac address and hard coded mapping for sub strings.  
  
Take care not to change the services/characteristic between mappings/changes, as most servers cache the setup to save on battery and reconnection time. Seen funny bugs when they change and the client device is not deleted and re-paired. 

Could also use something like SDL GameControllerMapping https://wiki.libsdl.org/SDL3/SDL_GameControllerMapping, if your target application is SDL based.

Possible a 3rd party tool that does something similar on Android.

closed 2023-12-25 by Kashyapkoshti