lemmingDev/ESP32-BLE-Gamepad #159: win10 cannot find the esp32 ble gamepad

jinshsa, 2023-01-15:
i upload the code to my esp32, my andriod phone can find the "esp32 ble gamepad" and connect to it, but my PC with a bluetooth adapter cann't find it , while my PC can find my phone and send file with my phone.

jinshsa, 2023-01-17:
> Would need to confirm if the BlueTooth adapter is BlueTooth Low Energy(BLE) supported. Not sure if you can confirm this in Windows. Search in the Windows Device Manager for `Microsoft Bluetooth LE Enumerator`. Should be installed and enabled.

Thanks, I should to try this way

lemmingDev, 2023-01-17:
[image]

In device manager you should see this one

lemmingDev, 2023-01-27:
How did you go?

VoicesJr, 2023-03-05:
I'm having the same issue. But the Bluetooth LE Enumerator driver is installed and enabled.

lemmingDev, 2023-03-05:
What version of the library are you using?
What code are you trying to run?

VoicesJr, 2023-03-09:
Hi 
I have 0.5.2 installed of the ESP-BLE-Gamepad. 
The Microsoft Bluetooth LE Enumerator has the Driver version 10.0.19041.488 if thats of any use. I've been trying the example PotAsAxis. It on macOS and I also got the serial print to start working when I connected the ESP32 to an Iphone via BT. 

Gratefull for any help!!

VoicesJr, 2023-03-12:
I have alsoe tried 0.5.0.

LeeNX, 2023-05-07:
Do any of the other ESP32 BLE examples work, like the ```BLE_notify```, I think you should be able to see this with ```Bluetooth LE Explorer``` on Windows and/or ```BLE Scanner (Connect & Notify)``` on Android.

lemmingDev, 2024-02-24:
No response from author - closed

closed 2024-02-24 by lemmingDev