lemmingDev/ESP32-BLE-Gamepad #160: Extended keypad 6x6

Auryza, 2023-01-27:
Hello, 

I'm a newbie, so I don't understand much yet.
I'm trying to expand matrix to 36 buttons. 6x6 . 6 Rows and 6 columns. 
I'm using code of keypad 4x4 .
But problem is, that after configuration i'm always see only 16 buttons in controller settings. 
What i'm doing wrong?  Some example of the code:

#define ROWS 6
#define COLS 6
uint8_t rowPins[ROWS] = {32, 33, 25, 26, 27, 14}; 
uint8_t colPins[COLS] = {4, 16, 17, 5, 18, 19}; 
uint8_t keymap[ROWS][COLS] =
    {
        {1, 2, 3, 4, 5, 6},    
        {7, 8, 9, 10, 11, 12},    
        {13, 14, 15, 16, 17, 18}, 
        {19, 20, 21, 22, 23, 24} 
        {25, 26, 27, 28, 29, 30}
        {31, 32, 33, 34, 35, 36}
};

lemmingDev, 2023-01-27:
The controller has 16 buttons by default, but supports up to 128

You have to tell it how many you want

```
BleGamepadConfiguration bleGamepadConfig;
bleGamepadConfig.setButtonCount(numOfButtons);
bleGamepad.begin(&bleGamepadConfig); 
```

Auryza, 2023-01-27:
Thank you, for your help.
Yes, is working now.

closed 2023-01-27 by Auryza