lemmingDev/ESP32-BLE-Gamepad #163: Is it posible to add next and previous track?

yorgios69, 2023-01-31:
Hi, i have been working with the gamepad library just clear, no issues, in the BLE Keyboar library there are the next and previous track functions but with the BLEKeyboard there are a lot of issues, so it its possible to add the next andd previous track to the BLEgamepad library?

LeeNX, 2023-05-05:
Should be possible, by adding additional ```Consumer Page``` entries, something similar to https://e2e.ti.com/support/wireless-connectivity/bluetooth-group/bluetooth/f/bluetooth-forum/450402/hid-consumer-page-report-descriptor-compatibility-issues
```
0x09, 0xb3, //       USAGE (Fast Forward)        // does scan next track on osx, but otherwise works great
0x09, 0xb4, //       USAGE (Rewind)              // does scan previous track on osx, but otherwise works great
0x09, 0xb5, //       USAGE (Scan Next Track)     // works great in windows & osx
0x09, 0xb6, //       USAGE (Scan Previous Track) // works great in windows & osx
```
Quick hack, would be to change the HID for Volume Inc/Dec - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/756dc28cb17f23bbbed18a136ed43cc814cb50b8/BleGamepad.cpp#L283
  
Another option would be to make it configurable, but I am then thinking how much of the HID table should we include? Does including all these options make the library any better or possible slower? That might be another issue.

lemmingDev, 2025-01-12:
Might close this as with https://github.com/Mystfit/ESP32-BLE-CompositeHID, you can create a composite gamepad / keyboard device, and the keyboard can control all the media keys

closed 2025-01-12 by lemmingDev